gnuradio/gnuradio #2764: QPSK constellation object not scaled correctly

jdemel, 2019-08-26:
If you use a `Constellation Object` with QPSK modulation, the symbols are not scaled properly, impling `|s|^2 != 1`. 

In `gr-digital/lib/constellation.cc` the QPSK symbols are defined as 
`gr_complex(-GR_M_SQRT2, -GR_M_SQRT2)`. 
They should be defined as 
`gr_complex(-1.0 / GR_M_SQRT2, -1.0 / GR_M_SQRT2)`.

All other constellations seem to stick to the `|s|^2 == 1` convention except for QPSK.

marcusmueller, 2024-01-07:
Still the case:

```c++
constellation_qpsk::constellation_qpsk()
{
    d_constellation.resize(4);
    // Gray-coded
    d_constellation[0] = gr_complex(-GR_M_SQRT2, -GR_M_SQRT2);
    d_constellation[1] = gr_complex(GR_M_SQRT2, -GR_M_SQRT2);
    d_constellation[2] = gr_complex(-GR_M_SQRT2, GR_M_SQRT2);
    d_constellation[3] = gr_complex(GR_M_SQRT2, GR_M_SQRT2);

    /*
      d_constellation[0] = gr_complex(GR_M_SQRT2, GR_M_SQRT2);
      d_constellation[1] = gr_complex(-GR_M_SQRT2, GR_M_SQRT2);
      d_constellation[2] = gr_complex(GR_M_SQRT2, -GR_M_SQRT2);
      d_constellation[3] = gr_complex(GR_M_SQRT2, -GR_M_SQRT2);
    */

    d_pre_diff_code.resize(4);
    d_pre_diff_code[0] = 0x0;
    d_pre_diff_code[1] = 0x2;
    d_pre_diff_code[2] = 0x3;
    d_pre_diff_code[3] = 0x1;

    d_rotational_symmetry = 4;
    d_dimensionality = 1;
    // normalization_t norm = POWER_NORMALIZATION;
    // normalize(norm);
    calc_arity();
}
```

However, fixing this will require an API break, as we reduce things by 6 dB…

So, doesn't look like we could backport a fix to this for 3.10, and I don't know whether it's worth fixing on main, then. @mormj , @willcode ?

willcode, 2024-01-08:
Also dqpsk.

Commented out code in qpsk and dqpsk (added in #6748 but never uncommented) would POWER_NORMALIZATION. In the PR, the author said this broke other stuff (equalizers, QA).