gnuradio/gnuradio #2766: QT GUI Tab Widget does not show Id field

jdemel, 2019-08-27:
If you want to organize your Qt GUI you may want to use the `QT GUI Tab Widget`. Other QT GUI blocks are placed inside this Widget with  `tab_widget_id@<tabnumber>: 0, 2, 2, 2` (values are examples) in the `GUI Hint` field.

Just like any other `Variable` block, the QT GUI Tab Widget needs to show the `Id` field. Otherwise, you'd need to guess from the generated code tab what the actual ID is and you cannot alter it. Which may place a significant burden on users with those auto-generated IDs.

At this point, I wonder how the decision to hide the ID field by default was made. This does make the function probe mostly useless or very complicated to use. Also, issues like the one described above pop up.

dl1ksv, 2019-08-27:
I plead to show the Id field for all blocks as in older versions. I see no benefit tho show the id only for special blocks.
By the way the option block shows the id, too.

dl1ksv, 2019-08-27:
Looking into the code , I saw that it's possible to conigure the visibility of the id in the yml file.
I did this for the Tab Widget #2767

jdemel, 2019-08-29:
@dl1ksv Thanks for taking care of this.

ThomasHabets, 2019-11-06:
`Probe signal` also needs to make this visible.

haakov, 2020-01-26:
@dl1ksv's PRs (for `master` and `maint-3.8`) have been merged, closing.

closed 2020-01-26 by haakov