gnuradio/gnuradio #2780: grc: add ability to force blocks to show id

mormj, 2019-08-30:
There are some blocks where the show_id field is not set, nor is it desired to be set, but I might want to see/change the block id in GRC. Could be for various reasons

1) clean up the automatic naming so that the generated .py is cleaner
2) reference the block by id from some other part of the flowgraph
3) the show_id didn't get set properly in the yml, and as a workaround I can change it, not wait until it is fixed.

This could be a context menu --> show block id

Thoughts?

jdemel, 2019-09-05:
I can see at least one use case for this. The function probe block needs to know block IDs of all blocks. If you want to generate a flowgraph that you want to embed in a project, it might be really useful to be able to set those block IDs. Though, you break encapsulation in this case.

Logismos, 2019-09-05:
As a new user, I've come across this issue while following the tutorials. The tutorial has us place a probe signal and a function probe block. Without an ID for the probe signal, I can't get the function probe to work in grc. @mormj, what was the workaround you used?

mormj, 2019-09-05:
There are 2 possible workarounds currently:
1) add the "show_id" into the yml for the block you need the id for
    -- have to go into the dir where the grc blocks are installed, modify, then relaunch grc
2) generate the flowgraph and note what the block id is

What I'm suggesting is we do one of the following:
1) have a way to display it in grc when it is hidden (right click --> show id)
    -- feature would need to be added
2) just show id for every block by default (pre-3.8 behavior)
   -- decision would need to be made to revert this behavior

jdemel, 2019-09-05:
I'd argue that it is not a good a idea to revert this behavior in 3.8. The decision was made for better or worse. But it would be a good enhancement to add an 'always show block ID' Option in GRC. Just like those snap to grid or hide port ID options.
This might even be easier than changing this on a block by block basis.

mormj, 2019-09-05:
I like that suggestion - will give that a go and see what folks think

PR #2795 (2019-09-17): grc: option toggle the show id on all blocks
Addresses the issue of losing access to the block id on many/most blocks
though there are scenarios where you may need to see them in GRC

Added a menu item tied into the grc prefs that will show the block ids

Fixes #2780
commit 44a6f5dd (2019-09-17): grc: option toggle the show id on all blocks

Addresses the issue of losing access to the block id on many/most blocks
though there are scenarios where you may need to see them in GRC

Added a menu item tied into the grc prefs that will show the block ids

Fixes #2780
commit 30717ed4 (2020-03-13): grc: option toggle the show id on all blocks

Addresses the issue of losing access to the block id on many/most blocks
though there are scenarios where you may need to see them in GRC

Added a menu item tied into the grc prefs that will show the block ids

Fixes #2780
closed 2019-09-17 by mbr0wn