gnuradio/gnuradio #2782: floflow gw grah generation with python 3 still imports python2 libraries

mugpahug, 2019-09-03:
Generating a flow graph that uses "Variable Config" and "XMLRPC Server" in GNU radio 3.8.0.0 (Python 3.7.4) still loads ConfigParser and SimpleXMLRPCServer, which is replaced by configparser and xmlrpc.server in python 3.

output:
`Generating: '/home/epeters/src/adfags-modem/src/GRC/zmq_file_single.py'

Generating: '/home/epeters/src/adfags-modem/src/GRC/zmq_file_single.py'

Executing: /usr/bin/python3 -u /home/epeters/src/adfags-modem/src/GRC/zmq_file_single.py

Traceback (most recent call last):
  File "/home/epeters/src/adfags-modem/src/GRC/zmq_file_single.py", line 21, in <module>
    import ConfigParser
ModuleNotFoundError: No module named 'ConfigParser'

>>> Done (return code 1)
`

after modifying the "import ConfigParser" to "import configparser as ConfigParser" in the output file SimpleXMLRPCServer also throws errors:

`Traceback (most recent call last):
  File "radioController.py", line 168, in <module>
    from zmq_file_single import zmq_file_single as GRC
  File "/home/epeters/src/adfags-modem/src/GRC/zmq_file_single.py", line 22, in <module>
    import SimpleXMLRPCServer
ModuleNotFoundError: No module named 'SimpleXMLRPCServer'
`

mugpahug, 2019-09-04:
In addition I found the following issues:
`def __init__(self, cfg=default.ini):` has to be ` def __init__(self, cfg='default.ini'):`

all `config.read('cfg')` have to be `config.read(cfg)`

Running the following command line against the python script generated by GNU radio mitigates XMLRPC and ConfigParser errors, as well as a the above mentioned issues with variable naming

`sed -i -e "s/'cfg'/cfg/g" -e "s/import ConfigParser/import configparser as ConfigParser/g" -e "s/import SimpleXMLRPCServer/import xmlrpc.server as SimpleXMLRPCServer/g" -E -e "s/cfg=(.*)\)/cfg='\1'\)/g" top_block.py
`

I tested this successfully on my python files generated using GNU radio 3.8.0.0

bidimensional, 2020-04-16:
I have the same issue.. gnuradio 3.8 import ConfigParser using Python 3, but in Python 3 the same library is called configparser. If i change the script manually it does work, but of course every time gnuradio generates the scripts it replaces with the wrong lib.

dkozel, 2020-04-17:
This needs to be fixed by conditionally importing the correctly named config parser library depending on the Python version. Something like the following. This is a bug fix only for the `maint-3.8` branch since `master` was fixed in October and is Python 3 only.

```
import sys
if sys.version_info[0] >= 3:
    import configparser
else:
    import ConfigParser as configparser
```

marcusmueller, 2024-01-07:
fixed in 3.8.2.0 and obsoleted by 3.9.0.0

PR #3429 (2020-05-22): grc: fix importing configparser in variable config block
Partially fixes #2782.

This backports #2867 to GNU Radio 3.8 so that the Variable Config block works in both Python 2 and Python 3.
closed 2024-01-07 by marcusmueller