gnuradio/gnuradio #2785: rpcbasic_sptr not defined in basic_block.h if GR_CTRLPORT is not defined either

carlesfernandez, 2019-09-03:
A recent commit seems to break [gnss-sdr](https://github.com/gnss-sdr/gnss-sdr) building against latest GNU Radio master and maint-3.8 branches.

The cause is here: https://github.com/gnuradio/gnuradio/commit/4c360371bb3f34401562e7cfcd66d8e774796091#r34938539

The obvious change seems 
```
#ifdef GR_CTRLPORT
std::vector<rpcbasic_sptr> d_rpc_vars; // container for all RPC variables
#endif 
```
but I don't know if this can have other side effects.

michaelld, 2019-09-04:
@carlesfernandez if you remove the `#ifdef GR_CTRLPORT` from around `#include <gnuradio/rpcregisterhelpers.h>`, does that work? That's what's done in `pycallback_object.h`, and may be what should be done here.

carlesfernandez, 2019-09-04:
Hi @michaelld 

yes, if I remove the `#ifdef GR_CTRLPORT` from around `#include <gnuradio/rpcregisterhelpers.h>`everything seems to work well.

May I do a PR for that? It should be against `maint-3.8`, isn't it?

michaelld, 2019-09-04:
@carlesfernandez I think that's the correct solution. Please do a PR on it against maint-3.8 and master. Looks like maint-3.7 doesn't need it, since it uses a Boost class instead.

carlesfernandez, 2019-09-04:
Done against maint-3.8 in #2787

carlesfernandez, 2019-09-20:
Replaced by #2805

carlesfernandez, 2019-09-24:
Fixed in 00c1b64fd974621c28b6f75b369f0aebc1ecc07f

closed 2019-09-24 by carlesfernandez

michaelld, 2019-09-25:
Thanks @carlesfernandez ... I had forgotten about this issue!