gnuradio/gnuradio #2793: pybombs install fails for 3.8 on windows.h

cclamb, 2019-09-07:
Installing gnuradio38 on ubuntu 19.04. Prior to this, I needed to manually install (via PIP, and conda respectively) mako and numpy (which I'm not sure if I should need to do?). Finally fails trying to find windows.h. Build sequence:

$ pybombs auto-config
$ pybombs recipes add-defaults
$ pybombs prefix init ~/gnuradio_3.8 -a gr38
$ pybombs -p gr38 install gnuradio38

I am executing this in a conda environment configured with pip and python3:

$ python -V
Python 3.7.4
$ pip -V
8.3.0

Final failure message:
$ tail /home/cclamb/gnuradio_3.8/src/gnuradio38/build/CMakeFiles/CMakeError.log                                  
/usr/bin/c++    -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized    -o CMakeFiles/cmTC_9c5c4.dir/CheckIncludeFile.cxx.o -c /home/cclamb/gnuradio_3.8/src/gnuradio38/build/CMakeFiles/CMakeTmp/CheckIncludeFile.cxx
/home/cclamb/gnuradio_3.8/src/gnuradio38/build/CMakeFiles/CMakeTmp/CheckIncludeFile.cxx:1:10: fatal error: windows.h: No such file or directory
 #include <windows.h>
          ^~~~~~~~~~~
compilation terminated.
make[1]: *** [CMakeFiles/cmTC_9c5c4.dir/build.make:66: CMakeFiles/cmTC_9c5c4.dir/CheckIncludeFile.cxx.o] Error 1
make[1]: Leaving directory '/home/cclamb/gnuradio_3.8/src/gnuradio38/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_9c5c4/fast] Error 2

velichkov, 2019-09-07:
Hi @cclamb,

It's pretty normal CMakeError.log to contains a lot of error messages. I've successfully built gnuradio 3.8 using pybombs and also have the same error in CMakeError.log. Better provide the output from the terminal.

See also https://github.com/gnuradio/gr-recipes/pull/155

cclamb, 2019-09-27:
Just installed 3.8 via pybombs, this was not an issue this time (found another one though, I'll put that in another item).

closed 2019-09-27 by cclamb