gnuradio/gnuradio #2802: FFT Filter Block corrupts output when using more than one thread

CharlesDove, 2019-09-15:
Good afternoon!  I'm working on a system to transmit ATSC signals, which includes a FFT Filter Block.  Since the process for producing the ATSC signal is computationally intensive, I'd like to use the "Num. Threads" option included with the FFT Filter Block to increase the number of threads.  When I increase the number of threads beyond 1, however, the output is seemingly corrupted and will not play on the connected TV.  The system works correctly on 1 thread.  
I'm using the Hackrf One and GNU Radio 3.8.0.0 on Manjaro Linux.  
My processor is: "8 × Intel® Core™ i7-6820HQ CPU @ 2.70GHz"
I've attached the flowgraph in question. Any idea what might be causing this?  Any help is appreciated!
[atsc-blade.zip](https://github.com/gnuradio/gnuradio/files/3614177/atsc-blade.zip)

drmpeg, 2019-09-16:
The number of threads option has been broken for a long time (years).

BTW, you can use the ATSC transmitter flow graph in <install_dir>/share/gnuradio/examples/dtv/uhd_atsc_tx.grc. Just replace the UHD sink with the osmosdr sink.

Some of the other blocks have been changed to increase performance. Unfortunately, the FFT filter block does take the most cycles.

CharlesDove, 2019-09-17:
Ron,
Thanks for the info!  I tried the flow graph you mentioned, but it unfortunately didn't resolve the bottlenecking I've been experiencing while modulating the atsc.  (The program can't keep up with the video stream, causing large amounts of tearing.  The program only uses 30% of available CPU, so the bottleneck is presumably due to python.)  Any advice on how to get around that?

michaelld, 2019-09-17:
Just because it's taking 30% of your total CPU doesn't mean one single thread isn't causing the issue. If you run "top" you'll likely see one thread/core maxed out.

The FFT filter is expensive, especially with 1 thread. The "multiply by sine" is also expensive ... you might consider the "FIR Xlating" Filter instead ... should be much more efficient.

drmpeg, 2019-09-17:
An i7-6820HQ should be plenty of CPU. It has a Passmark rating of 8780. I'm using an E5-1607 here, which has a Passmark rating of 5893 with no problems. Here's what my "top -H" looks like. Not much of a CPU load at all.

[image]

I'm guessing you have some other problem, but I'm not sure what. Be aware that the Transport Stream file has to be constructed properly. It looks like you're using the Adventure Time test clip, so you should be okay. If you're using some other clip, that could be the problem.

michaelld, 2019-09-17:
OK interesting. It's also possible that the CPU just can't process more data than it is already without more FFT filter threads (which don't work). The block/thread is at max CPU for what it can do, which doesn't mean 100% of 1 core even. It's possible.

CharlesDove, 2019-09-17:
Thanks for the help, guys.  I was actually having trouble with the "rotator" block that replaces the "multiply by sine," in the UHD example flow graph mentioned earlier, so I'm still using the "multiply by sine."  It sounds like that's contributing to the problem, so once I get back to my radio I'll check if that was user error+try the Xlating Filter+check thread usage and update you.

drmpeg, 2019-09-17:
The Xlating FIR filter is definitely the wrong way to go. It pegs the CPU. The FFT filter is by far the most performant for the RRC filter.

[image]

Looks like there's a bug with the rotator block in 3.8.

https://github.com/gnuradio/gnuradio/issues/2748

michaelld, 2019-09-17:
Holy moly that's ridiculous CPU usage for the Xlating block! I had no idea how inefficient it is! Too bad about the rotator being borked in 3.8 ... time to get it fixed I guess!

argilo, 2019-09-18:
A fix is available for the broken rotator block: https://github.com/gnuradio/volk/pull/278. It just needs to be merged into volk and then picked up GNU Radio.

jdemel, 2019-09-18:
A quick fix is: 
1. Run `volk_profile`
2. In `~/.volk/volk_config` edit the line `volk_32fc_s32fc_x2_rotator_32fc`
3. Remove any `_fma` extension. Thus, do not use the avx_fma kernel because that's the broken one .  VOLK defaults to the `avx_fma` kernel if `~/.volk/volk_config` does not exist or does not have an entry for `volk_32fc_s32fc_x2_rotator_32fc`.

The `avx_fma` kernel seems to be slightly faster on some machines, e.g. on the laptops in my lab but not on my desktop. For this specific kernel it is a bad idea to use the generic kernel performance-wise. I measure a ~30x speed-up between the `generic` and `u_avx` kernels.

CharlesDove, 2019-09-18:
Got the flow graph with the rotator working with that fix, thanks jdemel!  The playback now works flawlessly for exactly one playback, but if I try to repeat it in any way, the second and subsequent runs are all visually corrupted.  Restarting gnuradio companion or unplugging/replugging the radio doesn't appear to help, but restarting the computer resets the process, working perfectly for a single run then corrupted afterward.  Any ideas what might be causing that?

CharlesDove, 2019-09-20:
Update:  Testing with a shorter video clip, the clip can actually repeat several times before the video becomes visually corrupted.  In that case I suppose it's due to some sort of accumulating load that pushes one of the threads over its max.  Any ideas how to fix that?  
Alternatively, would implementing the process in C++ help? The only blocks that appear to have only Python implementations are the ATSC blocks, and I could look into porting those to C++.

drmpeg, 2019-09-20:
All the blocks are already written in C++. It's difficult for me to believe it's a performance issue. Compared to the DVB-T2 transmitter, the ATSC transmitter is light weight.

The cable QAM transmitter uses even less CPU. You may want to give that a try if you're using a cable ready TV. It's in <install_dir>/share/gnuradio/examples/dtv/catv_tx_64qam.grc. There's a different test .TS file to download. See the README.catv file.

marcusmueller, 2019-09-29:
So, there's quite a bit for me to unpack in here.

1. "The number of threads option has been broken for a long time (years).": @drmpeg I didn't know that. As far as I can tell, there's no issue here documenting that. I can't reproduce it; can you shed some more light on this, please? This **deeply** worries me.
2. "Freq X-Lating is slow": oha. this shouldn't be the case. I'll need to investigate with `perf`.
3. "In that case I suppose it's due to some sort of accumulating load that pushes one of the threads over its max. Any ideas how to fix that?" @CharlesDove I'm not convinced this has anything to do with load; how do you come to that conclusion?

drmpeg, 2019-09-29:
1) The number of threads option doesn't affect the output of the FFT filter, but it does ruin the performance.

Here's top -H with the DVB-S2 transmitter with 1 thread.

[image]

And with 2 threads. Not only is the FFT filter affected, but the python3.6 thread is using huge CPU. I'm not sure why the python3.6 thread goes from essentially zero to almost pinned.

[image]

2) I was measuring the frequency translating FIR filter.

marcusmueller, 2019-09-29:
@drmpeg  huh! I'll really have to look into that; for this instant, I'm relieved to hear that what the title of this issue says doesn't seem to be true. My admittedly wild guess / mental process regarding the python3.6 thread here goes like this:

* all these block threads are actually run under the python3.6 process "umbrella"; it's just that top shows the block name as thread name for these threads that declared a name
* the main thread is, guessing from the PIDs, however truly the second-from-top python3.6 thread. Maybe that thread simply carries the burden of communicating results in a scenario where distributing calculations among threads doesn't pay because the partial problems become smaller than the communication overhead
* We can debug that! 

So, would you mind running `perf top -a` (you might need to allow that, first, via `sudo sysctl -w kernel.perf_event_paranoid=-1`)? That should give you a system-overall view on how much time CPUs spend in which function (it's sampling of instruction pointers, so, Nyquist applies, but it's still very helpful).

If that yields significant amount of time in some strangely named fftw function, than the "this is just FFTw communication overhead" is right. If not, we might need to dig deeper by 

1. start the flowgraph, run `top` (or `htop`) to find the thread ID of that `python3.6` thread
2. `cd /some/tmp/folder; perf record -t ${that thread ID}` for a few seconds, end recording with ctrl-c
3. `perf report` from the same place later on (`perf top` is basically `record` & `report` rolled into a live view, which can be a bit bothersome for later analysis)

drmpeg, 2019-09-29:
You should be able to easily reproduce this. The test flow graph is in:

<install_dir>/share/gnuradio/examples/dtv/dvbs2_tx.grc

Just disable the UHD sink and QT GUI Frequency sink and add a throttle and null sink.

The test transport stream can be downloaded from here:

http://www.w6rz.net/adv16apsk910.ts

[image]

willcode, 2020-06-28:
In a very quick test - threading has some effect if output buffers and tap lengths are large. Breaking up a relatively small FFT doesn't help performance, and can actually hurt it due to threading overhead.

SteveOPLM, 2025-01-20:
> The number of threads option has been broken for a long time (years).
>
> BTW, you can use the ATSC transmitter flow graph in <install_dir>/share/gnuradio/examples/dtv/uhd_atsc_tx.grc. Just replace the [...]