gnuradio/gnuradio #2812: gr_modtool newmod path error on GR 3.8 pybombs install

ghostop14, 2019-09-24:
After installing GNU Radio 3.8 into a prefix via pybombs, the path created in the .gnuradio/config.conf is missing a "templates" portion of the path for 3.8.  The result is that the command "gr_modtool newmod" throws a "ModToolException: Could not find gr-newmod source dir." error.

I found the source of the issue.... the config.conf path generated during pybombs install looks like this:
newmod_path = <install prefix>/share/gnuradio/modtool/gr-newmod
but it should look like this:
newmod_path = <install prefix>/share/gnuradio/modtool/**templates**/gr-newmod

Updating config.conf manually by adding the missing /templates part fixes the issue.

primercuervo, 2019-10-04:
hi @ghostop14,

where is the absolute path of the `gnuradio/config.conf` file that you are mentioning? If it is `/home/user/.gnuradio/config.conf`, there is a high chance that this file was there _before_ your gnuradio 3.8 installation via PyBOMBS.

The logic behind my reasoning follows this [documentation link](https://wiki.gnuradio.org/index.php/Configuration_Files). The configuration file for the modtool newmod is located in `{prefix}/etx/gnuradio/conf.d/` and it's called `modtool.conf`, where the path should be correct (or at least it appeared to be so the last time I checked).

Please have a look at the contents of your prefix in order to confirm/deny the error. Keep in mind that a configuration file in `~/.gnuradio/config.conf` will override the settings of the prefixes for the user in that given machine.

ghostop14, 2019-10-06:
This was the one system I "upgraded".  Although I blew away the prefix and started from scratch.  I don't think I deleted $HOME/.gnuradio configs.  Maybe a solution could be for pybombs to check if it exists and ask the user to delete it and create a new one, back it up, and/or keep it?

marcusmueller, 2024-01-07:
this is a 3.7->3.8 upgrade problem, hence closing it. Sorry!

closed 2024-01-07 by marcusmueller