gnuradio/gnuradio #2813: PyGTKDeprecationWarning

duggabe, 2019-09-24:
When I load a new Embedded Python Block in a flowgraph, I get the
following warnings. This is before I have made any changes to the code.
I am using GR 3.8-maint, GRC 3.9.
```
/usr/local/lib/python3.7/dist-packages/gnuradio/grc/gui/Dialogs.py:362:
PyGTKDeprecationWarning: Using positional arguments with the GObject
constructor has been deprecated. Please specify keyword(s) for "title,
parent, action, buttons" or use a class specific constructor. See:
https://wiki.gnome.org/PyGObject/InitializerDeprecations
   transient_for=parent
/usr/local/lib/python3.7/dist-packages/gnuradio/grc/gui/Dialogs.py:362:
PyGTKDeprecationWarning: The "buttons" argument must be a
Gtk.ButtonsType enum value. Please use the "add_buttons" method for
adding buttons. See:
https://wiki.gnome.org/PyGObject/InitializerDeprecations
   transient_for=parent
/usr/lib/python3/dist-packages/gi/overrides/Gtk.py:571:
PyGTKDeprecationWarning: The keyword(s) "parent" have been deprecated in
favor of "transient_for" respectively. See:
https://wiki.gnome.org/PyGObject/InitializerDeprecations
   self._init(*args, **new_kwargs)
```

Notou, 2019-11-04:
I also get this warning.

duggabe, 2020-10-12:
I am currently running GR 3.8.1.0 and GRC 3.8.1.0
I have not encountered this error at all on this version.
Has it been fixed, or does it no longer apply?

duggabe, 2020-11-15:
With 3.9.0.0-git, the error no longer occurs.

closed 2020-11-15 by duggabe

willcode, 2020-11-15:
Also does not appear in 3.8.2.0.