gnuradio/gnuradio #2817: https://wiki.gnuradio.org/index.php/UbuntuInstall missing dependency for 18.04 Bionic Beaver

chrisjohgorman, 2019-09-26:
Hello,

the UbuntuInstall page says to install the following dependencies to build gnuradio 3.8 on ubuntu 18.04.

sudo apt install git cmake g++ libboost-all-dev libgmp-dev swig python3-numpy \
python3-mako python3-sphinx python3-lxml doxygen libfftw3-dev libcomedi-dev \
libsdl1.2-dev libgsl-dev libqwt-qt5-dev libqt5opengl5-dev python3-pyqt5 \
liblog4cpp5-dev libzmq3-dev python3-yaml python3-click python3-click-plugins 

The build works with this configuration, but make test fails for the zeromq tests.  Installing python3-zmq fixes the make test error.  Perhaps we should add python3-zmq to the install recipe?

Chris

777arc, 2019-10-01:
Thanks for finding that, the wiki is updated.

jdemel, 2020-01-15:
I suggest to close this issue because it is fixed.

closed 2020-01-15 by michaelld