gnuradio/gnuradio #2825: add_const_ff: not VOLKized

marcusmueller, 2019-09-28:
I was surprised to find that instead of an appropriate VOLK this is what `add_const_ff` does:

```
    while (size >= 8) {
        *optr++ = *iptr++ + d_k;
        *optr++ = *iptr++ + d_k;
        *optr++ = *iptr++ + d_k;
        *optr++ = *iptr++ + d_k;
        *optr++ = *iptr++ + d_k;
        *optr++ = *iptr++ + d_k;
        *optr++ = *iptr++ + d_k;
        *optr++ = *iptr++ + d_k;
        size -= 8;
    }

    while (size-- > 0) {
        *optr++ = *iptr++ + d_k;
    }
```

I mean, if we don't have a VOLK kernel for this yet, the second while loop does certainly constitute a good generic kernel. I have my doubts that this manual 8-vectorization actually helps any modern compiler be smart about auto-SIMDization; it's actually more likely to be harmful.

lemonpy, 2019-10-09:
I would like to work with this one. 
How big is size?(aprox).

lemonpy, 2019-10-10:
I did a small program to better understand of the output from add_const, using an small array initialized with 0's and d_k = 1.
The output is 1 0 1 0 1 0 1 0 ... . I'm missing something?

marcusmueller, 2019-10-13:
hi @lemonpy, nice having you work on this!

The whole point of this is that you'd just replace these two loops with a single call to [`volk_32f_x2_add_32f`](http://libvolk.org/doxygen/volk_32f_x2_add_32f.html); we already have an optimized function for this, is just not being used!

In GNU Radio,

> How big could be size?(aprox).
> I mean, I'm thinking in strip mining with an n=1024 but maybe it's too big for this problem (or not)

can't generally be answered; workload sizes can be anything from 1 item to half of your address space in items.

I don't understand your second question, sorry. Can you rephrase?

lemonpy, 2019-10-16:
hey @marcusmueller thanks for the reply, for the second question I did an small program with the while and a couple of arrays to see the output. For a zeroes array and a constant equals to 1, the output was 1010101... .
I did the changes, and I built it from source but I don't know how to check if its working properly.
I found some qa_gr_blocks tests but for tags and I didn't found for the calls to add function, can you point me some readings to find a proper way to check if the change is correct?

lemonpy, 2019-10-17:
also I found other similar occurrences of the same while that I'll check

grant-cox, 2019-11-12:
@lemonpy @marcusmueller is this issue still being worked on? I don't see an associated pull request for it, and it's been untouched for a while. If it's gone stale, I'll work on it.

rear1019, 2020-03-18:
Closing, fixed by PR #2917. We might want to back-port this to maint-3.8 and even maint-3.7.

closed 2020-03-18 by rear1019