gnuradio/gnuradio #2828: gr-blocks build failed on ubuntu 16.04 LTS

wilecoyote2019, 2019-09-29:
Hi,
I have troubles building gnuradio 3.7.13.4 on ubuntu 16.04. gr-blocks build fails :
[ 39%] Linking CXX shared library libgnuradio-blocks-3.7.13.4.so
[ 39%] Built target gnuradio-blocks
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

Here is the result of cmake : 

cmake -DCMAKE_BUILD_TYPE="Release" -DCMAKE_INSTALL_PREFIX=/opt/gnuradio-3.7.13.4 ../  
-- Build type set to Release.
-- Extracting version information from git describe...
-- Compiler Version: cc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O3 -DNDEBUG  -std=gnu99 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Building Static Libraries: OFF
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   chrono
--   atomic
-- Enabling use of known bad versions of Boost.
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable version "2.7.12", minimum required is "2.7") 
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.8.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Enabling python-support support.
--   Override with -DENABLE_PYTHON=ON/OFF
-- 
-- Configuring testing-support support...
--   Dependency CPPUNIT_FOUND = TRUE
--   Enabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
-- 
-- Configuring VOLK support...
-- Build type set to Release.
-- Extracting version information from git describe...
-- 
-- Python checking for python >= 2.7
-- Python checking for python >= 2.7 - found
-- 
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - found
-- 
-- Python checking for six - python 2 and 3 compatibility library
-- Python checking for six - python 2 and 3 compatibility library - found
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   system
-- Enabling use of known bad versions of Boost.
-- QA Testing is enabled.
--   Modify using: -DENABLE_TESTING=ON/OFF
[... 346 lines cut ...]
--   * gr-channels
--   * gr-noaa
--   * gr-pager
--   * gr-qtgui
--   * gr-trellis
--   * gr-uhd
--   * gr-utils
--   * gr-video-sdl
--   * gr-vocoder
--   * gr-fcd
--   * gr-wavelet
--   * gr-wxgui
--   * gr-zeromq
-- 
-- ######################################################
-- # Gnuradio disabled components                        
-- ######################################################
-- 
-- Using install prefix: /opt/gnuradio-3.7.13.4
-- Building for version: 3.7.13.4 / 3.7.13.4
-- Configuring done
-- Generating done
-- Build files have been written to: /home/oam/buildGunRadio/gnuradio/build

Any idea if I missed something ?

dkozel, 2019-09-29:
Somewhere higher on the build output there's likely to be the actual error, did you build with multiple threads? `make -j#` If you did then the error can occur long before the build finishes. Also you built into a custom prefix (`/opt/gnuradio-3.7.13.4`), is that in your linking paths?

Questions about building are better addressed on IRC/Slack or the mailing list, happy to help here.

wilecoyote2019, 2019-10-01:
Hi yes I was compiling with -j#. Compiling without -j# solved the problem, but compile take more time ^^ 
Thanks a lot!

closed 2019-10-01 by wilecoyote2019