gnuradio/gnuradio #2830: Pybombs not installing 3.8 properly on Ubuntu 18.4

mgarrett1955, 2019-10-01:
I am recieving the following error when trying to install gnuradio 3.8 on a fresh Ubuntu 18.4:

-- Python checking for numpy - not found
-- 
-- Configuring gnuradio-companion support...
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_PYTHON = ON
--   Dependency PYTHON_MIN_VER_FOUND = TRUE
--   Dependency PYYAML_FOUND = TRUE
--   Dependency MAKO_FOUND = TRUE
--   Dependency PYGI_FOUND = TRUE
--   Dependency GTK_GI_FOUND = TRUE
--   Dependency CAIRO_GI_FOUND = TRUE
--   Dependency PANGOCAIRO_GI_FOUND = TRUE
--   Dependency NUMPY_FOUND = FALSE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled gnuradio-companion but configuration checked failed
Call Stack (most recent call first):
  grc/CMakeLists.txt:95 (GR_REGISTER_COMPONENT)
[CMakeError.log](https://github.com/gnuradio/gnuradio/files/3676775/CMakeError.log)
[CMakeOutput.log](https://github.com/gnuradio/gnuradio/files/3676778/CMakeOutput.log)

mgarrett1955, 2019-10-01:
I located this comment in another similiar thread, but since Im using Pybombs, I do not know to to do this from the command line:

Note: to anyone reading this: ultimately fixed by forcing python version in cmake, ie:
cmake -DPYTHON_EXECUTABLE=/usr/bin/python ../
which picks up python 2.7 instead of python 3+, numpy is happy, lxml is happy, gnuradio-companion works...

mgarrett1955, 2019-10-01:
I could run this in /prefix/gnuradio/src/gnuradio/build and run cmake -DPYTHON_EXECUTABLE=/usr/bin/python ../, and then re run Pybombs?

Ru28, 2019-10-01:
Je reçois le message d'erreur suivant lorsque j'essaie d'installer gnuradio 3.8 sur une nouvelle Ubuntu 18.4:

- Vérification de python pour numpy - non trouvé
- Configuration du support de gnuradio-companion ...
- Dépendance ENABLE_GNURADIO_RUNTIME = ON
- Dépendance ENABLE_PYTHON = ON
- Dépendance PYTHON_MIN_VER_FOUND = TRUE
- Dépendance PYYAML_FOUND = TRUE
- Dépendance MAKO_FOUND = TRUE
- Dépendance PYGI_FOUND = TRUE
- Dépendance GTK_GI_FOUND = TRUE
- Dépendance CAIRO_GI_FOUND = TRUE
- Dépendance PANGOCAIRO_GI_FOUND = TRUE
- Dépendance NUMPY_FOUND = FALSE
Erreur CMake dans cmake / Modules / GrComponent.cmake: 75 (message):
gnuradio-companion activé par la force de l'utilisateur, mais la configuration vérifiée a échoué
Call Stack (l'appel le plus récent en premier):
grc / CMakeLists.txt: 95 (GR_REGISTER_COMPONENT)
CMakeError.log
CMakeOutput.log

mgarrett1955, 2019-10-02:
Should this be here, or should it be moved to Pybombs?

mormj, 2019-10-02:
Thanks for reporting - this should be moved over to pybombs to get proper attention.

mgarrett1955, 2019-10-02:
Thank you, I moved to Pybombs, ok to erase from here.

closed 2019-10-02 by mgarrett1955