gnuradio/gnuradio #2836: gr_modtool add `--argument-list' broken

kaslr, 2019-10-03:
The `--argument-list` parameter of `gr_modtool add` does not take effect, which I think is caused by 
https://github.com/gnuradio/gnuradio/blob/55c55fe3e17d556484ed9139e8e941a0da00f9c4/gr-utils/python/modtool/cli/add.py#L129-L130

This will always ask for prompt either set `argument-list` or not.

May be it should be
```
if not self.info['arglist']:
```
since the default value of `self.info['arglist']` is `""` instead of `None`:
https://github.com/gnuradio/gnuradio/blob/55c55fe3e17d556484ed9139e8e941a0da00f9c4/gr-utils/python/modtool/core/add.py#L54

primercuervo, 2019-10-03:
This is, most probably, dealt with these PR's:

master: https://github.com/gnuradio/gnuradio/pull/2826
maint-3.8: https://github.com/gnuradio/gnuradio/pull/2827

You might want to apply the patch locally to check it out before these are merged

marcusmueller, 2019-11-01:
Should work now. @kaslr please check with current master or maint-3.8!

closed 2019-11-01 by marcusmueller