gnuradio/gnuradio #2838: gr-osmocom not building with gnuradio 3.8

yaaseen, 2019-10-04:
I am trying to install gr-osmocom with gnuradio 3.8. gnuradio 3.8 was installed successfully in ubuntu 18.04 running in a virtual machine. 

`cmake -Wno-dev ../
-- Build type not specified: defaulting to release.
-- Extracting version information from git describe...
-- Configuring Boost C++ Libraries...
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   thread
--   system
--   chrono
--   date_time
--   atomic
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Checking for module 'gnuradio-iqbalance'
--   No package 'gnuradio-iqbalance' found
-- Could NOT find GNURADIO_IQBALANCE (missing: GNURADIO_IQBALANCE_LIBRARIES GNURADIO_IQBALANCE_INCLUDE_DIRS) 
-- Found gnuradio-uhd: /usr/local/include, /usr/local/lib/libgnuradio-uhd.so
-- Checking for module 'gnuradio-fcd'
--   No package 'gnuradio-fcd' found
-- gnuradio-fcd not found.
-- Could NOT find GNURADIO_FCD (missing: GNURADIO_FCD_LIBRARIES GNURADIO_FCD_INCLUDE_DIRS) 
-- Checking for module 'gnuradio-fcdproplus'
--   No package 'gnuradio-fcdproplus' found
-- gnuradio-fcdproplus not found.
-- Could NOT find GNURADIO_FCDPP (missing: GNURADIO_FCDPP_LIBRARIES GNURADIO_FCDPP_INCLUDE_DIRS) 
-- Checking for module 'libosmosdr'
--   No package 'libosmosdr' found
-- libosmosdr not found.
-- Checking for module 'librtlsdr'
--   No package 'librtlsdr' found
-- librtlsdr not found.
-- Checking for module 'libmirisdr'
--   No package 'libmirisdr' found
-- libmirisdr not found.
-- Checking for module 'libhackrf'
--   No package 'libhackrf' found
-- Could NOT find LIBHACKRF (missing: LIBHACKRF_LIBRARIES LIBHACKRF_INCLUDE_DIRS) 
-- Checking for module 'libairspy'
--   No package 'libairspy' found
-- Could NOT find LIBAIRSPY (missing: LIBAIRSPY_LIBRARIES LIBAIRSPY_INCLUDE_DIRS) 
-- Could NOT find SoapySDR (missing: SoapySDR_DIR)
-- Checking for module 'libfreesrp'
--   No package 'libfreesrp' found
-- libfreesrp not found.
CMake Error at CMakeLists.txt:179 (message):
  GnuRadio Runtime required to build gr-osmosdr

-- Configuring incomplete, errors occurred!
See also "/home/warmachine/repos/gr-osmosdr/build/CMakeFiles/CMakeOutput.log".
See also "/home/warmachine/repos/gr-osmosdr/build/CMakeFiles/CMakeError.log".
`

I am getting "GnuRadio Runtime required to build gr-osmosdr" error while cmake  in gr-osmocom. The other errors before that are just missing modules. I don't know why it is showing up, it had never shown in previous gnuradio versions. But even we ignore those, as it was fixed by installing appropriate modules.

After installing all the missing packages
`cmake -Wno-dev ../
-- Build type not specified: defaulting to release.
-- Extracting version information from git describe...
-- Configuring Boost C++ Libraries...
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   thread
--   system
--   chrono
--   date_time
--   atomic
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Checking for module 'gnuradio-iqbalance'
--   No package 'gnuradio-iqbalance' found
-- Could NOT find GNURADIO_IQBALANCE (missing: GNURADIO_IQBALANCE_LIBRARIES GNURADIO_IQBALANCE_INCLUDE_DIRS) 
-- Found gnuradio-uhd: /usr/local/include, /usr/local/lib/libgnuradio-uhd.so
-- Checking for module 'gnuradio-fcd'
--   No package 'gnuradio-fcd' found
-- gnuradio-fcd not found.
-- Could NOT find GNURADIO_FCD (missing: GNURADIO_FCD_LIBRARIES GNURADIO_FCD_INCLUDE_DIRS) 
-- Checking for module 'gnuradio-fcdproplus'
--   No package 'gnuradio-fcdproplus' found
-- gnuradio-fcdproplus not found.
-- Could NOT find GNURADIO_FCDPP (missing: GNURADIO_FCDPP_LIBRARIES GNURADIO_FCDPP_INCLUDE_DIRS) 
-- Checking for module 'libosmosdr'
--   Found libosmosdr, version 
-- Found libosmosdr: /usr/include, /usr/lib/x86_64-linux-gnu/libosmosdr.so
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 0.5.3
-- Found librtlsdr: /usr/include, /usr/lib/x86_64-linux-gnu/librtlsdr.so
-- Checking for module 'libmirisdr'
--   Found libmirisdr, version 0.0.0
-- Found libmirisdr: /usr/include, /usr/lib/x86_64-linux-gnu/libmirisdr.so
-- Checking for module 'libhackrf'
--   Found libhackrf, version 0.5
-- Found LIBHACKRF: /usr/lib/x86_64-linux-gnu/libhackrf.so  
-- Checking for module 'libairspy'
--   Found libairspy, version 1.0
-- Found LIBAIRSPY: /usr/lib/x86_64-linux-gnu/libairspy.so  
-- Could NOT find SoapySDR (missing: SoapySDR_DIR)
-- Checking for module 'libfreesrp'
--   No package 'libfreesrp' found
-- Found libfreesrp: /usr/include, /usr/lib/x86_64-linux-gnu/libfreesrp.so
CMake Error at CMakeLists.txt:179 (message):
  GnuRadio Runtime required to build gr-osmosdr

-- Configuring incomplete, errors occurred!
See also "/home/warmachine/repos/gr-osmosdr/build/CMakeFiles/CMakeOutput.log".
See also "/home/warmachine/repos/gr-osmosdr/build/CMakeFiles/CMakeError.log".
`

Does anyone knows why this happens? is this something I am doing wrong ?
Please let me know

michaelld, 2019-10-04:
If you got gr-osmosdr from its primary source, then that version isn't compatible with GR38. There are a couple unofficial versions that might work; I think this one is the most current / most likely: < https://github.com/igorauad/gr-osmosdr/tree/gr3.8 >.

dkozel, 2019-10-05:
`gr-osmocom` is looking for a new maintainer. Unfortunately until someone takes on that role the updated code isn't likely to be merged upstream. There's nothing that the main GNU Radio codebase can do to help at this point.
http://lists.osmocom.org/pipermail/osmocom-sdr/2019-September/001983.html

closed 2019-10-05 by dkozel

Ghaz75, 2019-12-30:
> `gr-osmocom` is looking for a new maintainer. Unfortunately until someone takes on that role the updated code isn't likely to be merged upstream. There's nothing that the main GNU Radio codebase can do to help at this point.
> http://lists.osmocom.org/pipermail/osmocom-sdr/2019-September/001983.html

Dear! I carried out  "make uninstall"  of the gr-osmosdr from main branch and used gr3.8 branch of your given link. Still gr-gsm is throwing mpir error .

velichkov, 2020-01-03:
Hi @Ghaz75,

> Still gr-gsm is throwing mpir error during cmake . .

Better report `gr-gsm` problems in the [gr-gsm repository](https://github.com/ptrkrysik/gr-gsm/) or in the [gr-gsm's mailing list](https://github.com/ptrkrysik/gr-gsm/#mailing-list)

Do you have `libgmp-dev` package installed? If not try installing it (`sudo apt-get install libgmp-dev`) and if it still fails provide the full cmake output in a new issue in the gr-gsm project.