gnuradio/gnuradio #2839: gnuradio/gr-uhd/lib/usrp_source_impl.cc:150:18: error

ibra1212, 2019-10-04:
Hi I have a problem when I do the 'make' for gnuradio installation. the standard output shows this:

gnuradio/gr-uhd/lib/usrp_source_impl.cc: In member function ‘virtual void gr::uhd::usrp_source_impl::set_rx_agc(bool, size_t)’:

/gnuradio/gr-uhd/lib/usrp_source_impl.cc:150:18: error: ‘class uhd::usrp::multi_usrp’ has no member named ‘set_rx_agc’
return _dev->set_rx_agc(enable, chan);

/gnuradio/gr-uhd/lib/usrp_source_impl.cc:150:41: error: return-statement with a value, in function returning 'void' [-fpermissive]
     return _dev->set_rx_agc(enable, chan);

make[2]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_source_impl.cc.o] Error 1
make[1]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/all] Error 2
make: *** [all] Error 2

Could you guide me if possible? Thank you in advance !

michaelld, 2019-10-04:
Clearly the UHD API isn't compatible with this GR source!

Which version of GR are you trying to build from source, and what version of UHD is currently installed?

ibra1212, 2019-10-04:
hi,
I installed the UHD from this link:
https://github.com/fairwaves/UHD-Fairwaves
which version can I install?

michaelld, 2019-10-04:
I'd recommend the latest release: 3.14.1.1, found from here: https://github.com/EttusResearch/uhd

ibra1212, 2019-10-04:
i am using the fairwaves UmTRx hardware to set up a 2G GSM network for a test do you think it can do it?

michaelld, 2019-10-04:
OK, well then you do need the UHD Fairwaves you note. Have you contacted Fairwaves about this issue? If not, I'd recommend doing so.

What version of GR are you trying to build?

ibra1212, 2019-10-04:
no i am not contacting.

but I use this gr:
https://github.com/osmocom/gr-osmosdr
no i am not contacting.

but I use this gr:
https://github.com/osmocom/gr-osmosdr

it is a tutorial that I am on the Umtrx website:
https://umtrx.org/journey-to-gsm-part-2-installation/

ibra1212, 2019-10-04:
no I did not contact them.

but I use this gr:
https://github.com/osmocom/gr-osmosdr

it is a tutorial that I am on the Umtrx website:

https://umtrx.org/journey-to-gsm-part-2-installation/

michaelld, 2019-10-04:
When you do `uhd_usrp_probe` or any other UHD command, what is the first line that's shown?

michaelld, 2019-10-04:
BTW> That guide is pretty old ... the "build-gnuradio" script is outdated ... which is probably why you're experiencing issues.

michaelld, 2019-10-04:
I'd recommend building GR from source instead of using the script. Once the dependencies are installed, it's pretty straight forward.

ibra1212, 2019-10-04:
root@openbts:~# uhd_find_devices 
linux; GNU C++ version 4.8.2; Boost_105400; UHD_003.008.004-0-unknown

--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:
    type: umtrx
    addr: 192.168.10.2
    name: UmTRX
    serial: 52

root@openbts:~# 

root@openbts:~# uhd_usrp_probe 
linux; GNU C++ version 4.8.2; Boost_105400; UHD_003.008.004-0-unknown

-- UmTRX driver version: 1.0.3-141-6c9567c-dirty
-- Opening a UmTRX device... 192.168.10.2
-- user_mtu.recv_mtu = 1472
-- user_mtu.send_mtu = 1472
-- fifo_ctrl.window_size = 15
-- TempA: 72.250000 C
-- Detected UmTRX 2.1
-- Diversity switch for channel 1: false
-- Diversity switch for channel 2: false
-- Known PA types: NONE EPA881F40A EPA942H40A EPA1800F37A 
-- Installed PA for sideA: NONE
-- Installed PA for sideB: NONE

UHD Error:
    Control packet attempt 0, sequence number 160:
    RuntimeError: no control response, possible packet loss

UHD Error:
    Control packet attempt 1, sequence number 161:
    RuntimeError: no control response, possible packet loss

UHD Error:
    Control packet attempt 2, sequence number 162:
    RuntimeError: no control response, possible packet loss
Error: RuntimeError: fifo ctrl timed out looking for acks

ibra1212, 2019-10-04:
OK well received !
A big thank you also for your help and advice.
May peace be with you !

michaelld, 2019-10-04:
You're welcome! Good luck. I'm closing this issue since it really has nothing to do with GR .... it's an issue with UHD-Fairwaves and/or build-gnuradio .

closed 2019-10-04 by michaelld