gnuradio/gnuradio #2849: Costas loop performances affected by output blocks

jybz, 2019-10-17:
Costas loop performances affected by output blocks 
Hello there !

I just want to report that the costas loop get disturbed my the output.

For example : (/!\ I just note that I forget to rename the constellation diagram)
[image]
By default, that schema works perfectly :
[image]
But, if I decrease the number of point in the time sink :
[image]
[image]
[image]
the costas loop get totally disturbed.
The worst is, that with that schema :
[image]
I even don't know what kind of workarround I can use.
[image]

Someone suggest that the µC (a i5-5300U) was overload :
<pre>
top - 10:28:02 up 26 days, 19:32, 15 users,  load average: 3,46, 1,43, 1,05
Tâches: 265 total,   3 en cours, 262 en veille,   0 arrêté,   0 zombi
%Cpu(s): 68,1 ut, 18,3 sy,  0,2 ni, 13,3 id,  0,1 wa,  0,0 hi,  0,0 si,  0,0 st
MiB Mem :   7875,7 total,    530,4 libr,   3977,4 util,   3368,0 tamp/cach
MiB Éch :   4096,0 total,    157,8 libr,   3938,2 util.    876,3 dispo Mem
  coordonnées de défilement: y= 1/265 (tâches), x = 1/12 (champs)
PID UTIL.     PR  NI    VIRT    RES    SHR S  %CPU  %MEM    TEMPS+ COM.
18526 jybz      20   0 1069008 170912  87020 S 142,9   2,1   1:18.39 python2
 6974 jybz      20   0 1070856  43704  13436 R  98,7   0,5  33:38.51 konsole
16842 jybz      20   0   18724    680    612 R  67,4   0,0   0:36.32 xxd
16163 root      20   0       0      0      0 I   8,6   0,0   0:03.71 kworker/u16:1-events_unbound
18545 root      20   0       0      0      0 I   8,6   0,0   0:04.52 kworker/u16:3-events_unbound
23169 root      20   0       0      0      0 I   8,6   0,0   0:08.63 kworker/u16:0-events_unbound
 6202 root      20   0 3012160 758212 728416 S   3,3   9,4 207:56.02 Xorg
 6684 jybz      20   0 3037712  72644  32680 S   2,7   0,9 255:15.29 kwin_x11
 6465 jybz      20   0  396988  33012  18120 S   2,0   0,4   1:53.63 konsole
 6803 jybz      20   0 3724076 259256  43664 S   0,7   3,2 118:22.19 plasmashell
 6847 jybz      20   0 5491948   7816   5108 S   0,7   0,1  96:41.53 pulseaudio
 2919 root      35  15   18248  14264   1332 S   0,3   0,2  10:28.59 preload
 4057 jybz      20   0 9491988 340992  46056 S   0,3   4,2  60:44.11 Web Content
 4078 jybz      20   0 2817300 243100  35352 S   0,3   3,0  33:46.66 Web Content
 4158 jybz      20   0 2783564 158168  64152 S   0,3   2,0  10:20.04 Web Content
 4174 jybz      20   0 2670556 165200  66388 S   0,3   2,0   8:45.11 Web Content
 4249 jybz      20   0 2657592 180376  56028 S   0,3   2,2  10:12.38 Web Content
 7733 jybz      20   0  342772  49280   7536 S   0,3   0,6  17:23.43 net_applet
12087 jybz      20   0 1313188 334576  18784 S   0,3   4,1  16:32.97 gnuradio-compan
    1 root      20   0  167332   6468   4084 S   0,0   0,1   0:13.62 systemd
[...]
</pre>
But even if it is fully load, I dont' understand why processing more data in output will be better for the costas loop rather processing less.

jybz, 2019-10-17:
Well, it seems that it might be related to the fifo file in the sink file.
Back on that schema :
[image]
I have this result :
[image]
and the top command :
<pre>
top - 12:42:46 up 26 days, 21:47, 15 users,  load average: 2,90, 1,57, 0,98
Tâches: 280 total,   1 en cours, 279 en veille,   0 arrêté,   0 zombie
%Cpu(s): 55,2 ut,  7,1 sy,  0,1 ni, 37,3 id,  0,2 wa,  0,0 hi,  0,1 si,  0,0 st
MiB Mem :   7875,7 total,    986,8 libr,   3886,3 util,   3002,7 tamp/cache
MiB Éch :   4096,0 total,   2367,3 libr,   1728,7 util.   1683,8 dispo Mem

  PID UTIL.     PR  NI    VIRT    RES    SHR S  %CPU  %MEM    TEMPS+ COM.
26557 jybz      20   0 1069180 170660  86976 S 233,6   2,1   4:02.91 python2
 6684 jybz      20   0 3031052  71548  32284 S   7,0   0,9 262:11.74 kwin_x11
 6202 root      20   0  487660 159452 128160 S   4,0   2,0 212:57.82 Xorg
21108 jybz      20   0 2302924 242876 107068 S   3,3   3,0   3:21.57 plasmashell
 6847 jybz      20   0 4967660   8808   6008 S   1,0   0,1  98:05.87 pulseaudio
26488 jybz      20   0   18876    756    684 S   1,0   0,0   0:01.03 cat
 4174 jybz      20   0 2659952 168468  72084 S   0,7   2,1  11:26.75 Web Content
 6974 jybz      20   0 1073696  66844  34972 S   0,7   0,8  33:53.94 konsole
 2919 root      35  15   18248  14264   1332 S   0,3   0,2  10:34.64 preload
 4008 jybz      20   0   20,6g 145504  51148 S   0,3   1,8  30:02.28 WebExtensions
 4078 jybz      20   0 2811540 256640  50732 S   0,3   3,2  34:09.39 Web Content
    1 root      20   0  167332   6468   4084 S   0,0   0,1   0:13.88 systemd
    2 root      20   0       0      0      0 S   0,0   0,0   0:00.17 kthreadd
[...]
</pre>

We can see that the python program consumes twice more as previous post.

The difference on the fifo itself was, the non-working :
`xxd ./fifo-file`
working :
`cat ./fifo-file >/dev/null`

For the file sink output, the "bug" might be considered as closed. The. For the time sink, I let you decided.

marcusmueller, 2024-01-07:
I'm having a hard time reproducing. My guess is that due to CPU load, the limeSDR driver had to drop samples, with catastrophic results for the signal.

closed 2024-01-07 by marcusmueller