gnuradio/gnuradio #2850: Crashing process with Qt time sink trigger delays

jybz, 2019-10-17:
Hello !

I note that I can crash with shifting the delay for the trigger, only few clics (mainly 3 clics).
For example, this this simple schema :
[image]
When I start it :
[image]
if I clic (generally between 2 and 5), on the delay for the trigger, it crash and close the window.
On the terminal, I have no information :
<pre>INFO: device_handler::enable_channels(): SISO CH0 set for device number 0.
INFO: device_handler::set_samp_rate(): set sampling rate: 13.4947 MS/s.
INFO: device_handler::set_rf_freq(): RF frequency set [RX]: 13.4947 MHz.
INFO: device_handler::set_analog_filter(): RX LPF configured
INFO: device_handler::set_gain(): set gain [RX] CH0: 70 dB.
INFO: device_handler::set_antenna(): Selected RX path: LNAW
CH0 antenna set [RX]: LNAW.
INFO: device_handler::calibrate(): Rx calibration finished
INFO: source_impl::init_stream(): source channel 0 (device nr. 0) stream setup done.

>>> Done
</pre>

marcusmueller, 2019-11-01:
hey, could you do the following:

1. generate the python file from the GRC file using the "generate" button
2. run that python file manually from console, check whether the crash still occurs, and if so, whether the terminal contains the same info
3. run it as `gdb --args python2 /path/to/your/python/flowgraph.py`, type `run` (enter), trigger the crash, and then type `backtrace`(enter) on the GDB command line.

Share the backtrace with us, please!

surfzoid, 2023-01-13:
> Hello !
> 
> I note that I can crash with shifting the delay for the trigger, only few clics (mainly 3 clics). For example, this this simple schema : [image] When I start it : [image] if I clic (generally between 2 and 5), on the delay for the trigger, it crash and close the window. On the terminal, I have no information :
> 
> INFO: device_handler::enable_channels(): SISO CH0 set for device number 0.
> INFO: device_handler::set_samp_rate(): set sampling rate: 13.4947 MS/s.
> INFO: device_handler::set_rf_freq(): RF frequency set [RX]: 13.4947 MHz.
> INFO: device_handler::set_analog_filter(): RX LPF configured
> INFO: device_handler::set_gain(): set gain [RX] CH0: 70 dB.
> INFO: device_handler::set_antenna(): Selected RX path: LNAW
> CH0 antenna set [RX]: LNAW.
> INFO: device_handler::calibrate(): Rx calibration finished
> INFO: source_impl::init_stream(): source channel 0 (device nr. 0) stream setup done.
> 
> >>> Done

Hello jybz

surfzoid, 2023-01-13:
Hello Marcus, jybz and us work on a git lab project for mageia and mlo, jybz loose his github login info special the email adress, i guest throught this discution you will have his mail and can help us, the git lab project https://gitlab.com/mageialinux-online.org/magilaro and also our work for mlo team https://www.mageialinux-online.org/repository/8/x86_64/media/core/  thanks by advance

marcusmueller, 2024-01-07:
Can't reproduce

closed 2024-01-07 by marcusmueller