gnuradio/gnuradio #2851: gr-uhd grc 3.8 Unable to run uhd_fft sample

ernestp, 2019-10-18:
Using uhd_fft file (https://github.com/gnuradio/gnuradio/blob/master/gr-uhd/examples/grc/uhd_fft.grc)

Inside GRC it's look like error pointing at antenna=RX2, perhaps need quotes around RX2

Loading: "/GnuRadio/uhd_fft_yaml.grc"
>>> Done

Generating: '/GnuRadio/uhd_fft.py'

Executing: /Library/Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Contents/MacOS/Python -u /GnuRadio/uhd_fft.py

  File "/GnuRadio/uhd_fft.py", line 45
    def __init__(self, antenna=RX2, args=, fft_size=1024, freq=2.45e9, gain=20, samp_rate=1e6, spec=, stream_args=, update_rate=.1, wire_format=):
                                         ^
SyntaxError: invalid syntax

>>> Done (return code 1)

michaelld, 2019-10-28:
Wondering if this was fixed in gr::master commit https://github.com/gnuradio/gnuradio/commit/c919c7c7d269ca9d0cd2fc13a3619f04239f9f26 ... might be worth trying if you can do so.

ernestp, 2019-10-29:
Thanks @michaelld, just not sure how. I'm using https://github.com/ktemkin/gnuradio-for-mac-without-macports/releases/tag/v3.8.0.0-pre1 which was released after specified change. Not sure how to check if this change was applied in build.

michaelld, 2019-10-29:
@ernestp OK yeah you can't really do much since you're using the pre-built binary. Hopefully @ktemkin will provide an update to the (sometime) forthcoming 3.8.1 release, which I think will contain a fix for this issue. This is a known issue that is being, or has already been, addresses ... so I'm going to go ahead and close this issue ticket. If you feel otherwise, please reopen & we'll hope that something happens so-as to resolve this issue for pre-built binary users such as yourself.

closed 2019-10-29 by michaelld