gnuradio/gnuradio #2852: "transcendental function" block has bad name; shouldn't exist

marcusmueller, 2019-10-19:
… it includes sqrt, which is an analytic function, not transcendental at all.

The real problem, though, is that it's a wrapper for a few functions from a specific library (cmath). Since the number of functions wrapped is limited to begin with (13, to be exact), and nearly all of these exist as VOLK kernels somehow, this block should actually rather be split – who's going to look for "transcendental function" when they want a sine, or an exponential function?

The functions wrapped aren't even named in the doxygen, and that is misleading, to top that off, because the cmath reference suggests that all `cmath` functions are wrapped.

marcusmueller, 2019-10-19:
A viable solution would be to kill the `#defines` and finally getting around writing templated blocks that take `float (*func)(float)`, or more generally `typename T`, `T (*func)(T)`, and maybe `T (*func)(std::complex<T>)`, `std::complex<T> (*func)(T)`, as well as the VOLK-typical `void (*func)(T1*, T2*, unsigned int)`.

aiicore, 2020-05-10:
I can take this issue, would be nice to get familiar with blocks and refresh rusty C++ knowledge. Transcendental block should be split into individual ~13 blocks under Math Operations category, preferably using VOLK kernels + documentation, right?

marcusmueller, 2024-01-07:
this was, looking back, a feature request, and as such is still valid.

aiicore, 2024-01-08:
@marcusmueller if CLA's are no longer a requirement, could someone drop `Needs CLA` tag from #3480 and maybe all of my other PR's? Then I'll reopen them.

willcode, 2024-01-08:
Right, CLAs are not required. If you reopen the PRs, we'll assume that you accept the DCO. 3480 already has a signoff. Just make sure the others have a git signoff.

aiicore, 2024-01-08:
@marcusmueller, Transcendental block should be cleared up (defines -> templates) and renamed to Library C function wrapper, or should I continue with splitting it to 13 independent blocks like in the old PR?