gnuradio/gnuradio #2856: GNU Radio 3.8 installed with a package manager hard codes GR_PYTHON_DIR

noc0lour, 2019-10-21:
This breaks OOT builds into custom prefixes since the `GR_PYTHON_DIR` is not user settable anymore. 
This can be fixed by adding a `CACHE STRING` in `GnuradioConfig.cmake.in`#L40 to have a `-GR_PYTHON_DIR` in the OOT cmake override the variable in the configuration.  
For the `PYTHON_EXECUTABLE` this is neither necessary nor advisable since the python version should be hardcoded due to swig compatibility issues across different versions. For the install directory it's not as bad though. 

@nickoe might have a more detailed bug report for reproduction.

nickoe, 2019-10-22:
I can test the issue easily with this oneliner:
```
gr_modtool newmod testoot && pushd gr-testoot && mkdir build_ninja && cd build_ninja && cmake ../ -DCMAKE_INSTALL_PREFIX=/tmp/nocolor_test -DCMAKE_BUILD_TYPE=Debug -DCMAKE_C_COMPILER=clang -DCMAKE_CXX_COMPILER=clang++ -DPYTHON_EXECUTABLE=/usr/bin/python3.7 -G Ninja && ninja install ; popd
```
It is not specific to the Ninja generator, it also fails with regular make.
```
gr_modtool newmod testoot && pushd gr-testoot && mkdir build && cd build && cmake ../ -DCMAKE_INSTALL_PREFIX=/tmp/nocolor_test && make install ; popd
```

A snippet of the log:
```
-- Installing: /tmp/nocolor_test/share/doc/gr-testoot/html/tab_a.png
-- Installing: /tmp/nocolor_test/share/doc/gr-testoot/html/dynsections.js
-- Installing: /tmp/nocolor_test/share/doc/gr-testoot/html/jquery.js
-- Installing: /tmp/nocolor_test/share/doc/gr-testoot/html/tabs.css
CMake Error at python/cmake_install.cmake:49 (file):
  file cannot create directory: /usr/lib/python3.7/site-packages/testoot.
  Maybe need administrative privileges.
Call Stack (most recent call first):
  cmake_install.cmake:51 (include)
```

I would have expected it to add the `${CMAKE_INSTALL_PREFIX}` before `/usr/lib/python3.7/site-packages`.

marcusmueller, 2019-11-01:
Thanks for this bug report. Anyone have a PR?

nickoe, 2019-11-08:
@marcusmueller Sorry, no, I never got around to that, I spent all my time for this round on finding the cause, I don't think anyone else proposed a fix in this direction yet.

ZangMT, 2019-12-19:
Maybe that is the proxychains bug.  I met the same problem when installed pybombs using proxychains and shadowsocks. 
REF https://www.jianshu.com/p/a8739782a801

nickoe, 2020-10-03:
This is still an issue with 3.8.1

willcode, 2021-03-22:
Fixed by https://github.com/gnuradio/gnuradio/pull/3944 then https://github.com/gnuradio/gnuradio/pull/4032, and will be in v3.8.3.0. Closing.

closed 2021-03-22 by willcode