gnuradio/gnuradio #2859: grc: show message ports option in blocks does nothing

mormj, 2019-10-24:
In GR 3.8, at least several blocks (copy, selector, qt gui frequency sink, probably more) with the show message ports option - the option has no effect.  (reported by @devnulling)

[image]

[image]

mormj, 2021-04-28:
Fixed on 3.9/master - still exists on 3.8 - need to find and backport the commit that fixed it

mormj, 2021-04-29:
I think it is this PR that fixed the issue (3.9 and beyond)
https://github.com/gnuradio/gnuradio/pull/3934

@willcode - good candidate for backport?

dkozel, 2021-09-29:
@willcode Was this backported?

willcode, 2021-09-29:
Already in 3.9 (#3934) and 3.8 (#4556).

closed 2021-09-29 by mormj