gnuradio/gnuradio #2861: Struct Variable fails with field0

cclamb, 2019-10-26:
When I create a struct variable in gnuradio-companion, I need to leave a field called 'field0' in the struct or it emits an assertion failure: 

Assertion "(str(field0) or "a").isalnum()" failed.

It seems the default name 'field0' is hardcoded.

cclamb, 2019-10-26:
...if you add new fields, and then change the name of the added field, you'll see this kind of behavior as well.

marcusmueller, 2019-11-01:
Thanks for reporting!

haakov, 2020-01-04:
Has this been fixed? I'm not able to reproduce this behaviour.

hackery, 2020-01-15:
I'm seeing it today on 3.7.11 (Ubuntu 18.04). Building 3.8.0 now here to try that.

haakov, 2020-01-15:
@cclamb Did you encounter this on 3.7 as well?

cclamb, 2020-01-29:
Yep

marcusmueller, 2024-01-07:
Seems solved:
[image]

https://gist.github.com/marcusmueller/f39b82dc44a4a8e82329094078904e64

closed 2024-01-07 by marcusmueller

jmfriedt, 2024-01-25:
If we add to your GRC flowchart a new variable using the Struct Variable as "variable_struct_0.eatmyshorts" then the error message 
```
	Value "variable_struct_0.eatmyshorts" cannot be evaluated:
	name 'variable_struct_0' is not defined
```
hints at the fact that the Struct Variable name was not added to the Python script namespace.