gnuradio/gnuradio #2864: help please. gnu-companion disabled

jin-github102, 2019-10-27:
compiling gnuradio, gnu-companion is disabled component. I think maybe it is because of PYTHONPATH. But i dont know how to deal with it.

would you please give me some instructions about it? thank you.

dell@dell-XPS-15-9550:~/gnuradio/build$ sudo cmake ../
-- Build type not specified: defaulting to release.
-- Build type set to Release.
-- Extracting version information from git describe...
-- Compiler Version: cc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Found Boost: /usr/include (found suitable version "1.58.0", minimum required is "1.53") found components:  date_time program_options filesystem system regex thread unit_test_framework chrono atomic 
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found suitable exact version "3.5.2") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found suitable version "3.5.2", minimum required is "2.7.6") 
-- Python checking for six - python 2 and 3 compatibility library - found
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.8.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Dependency SIX_FOUND = TRUE
--   Enabling python-support support.
--   Override with -DENABLE_PYTHON=ON/OFF
-- 
-- Configuring testing-support support...
--   Dependency Boost_FOUND = TRUE
--   Enabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
-- 
-- Configuring VOLK support...
-- Build type set to Release.
-- Extracting version information from git describe...
-- 
-- Python checking for python >= 2.7
-- Python checking for python >= 2.7 - found
-- 
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - found
-- 
-- Python checking for six - python 2 and 3 compatibility library
-- Python checking for six - python 2 and 3 compatibility library - found
-- Found Boost: /usr/include (found suitable version "1.58.0", minimum required is "1.35") found components:  filesystem system 
-- QA Testing is enabled.
--   Modify using: -DENABLE_TESTING=ON/OFF
-- System profiling is disabled.
--   Modify using: -DENABLE_PROFILING=ON/OFF
-- Compiler name: GNU
-- x86* CPU detected
-- Compiler doesn't support NEON, Overruled arch neon
-- Compiler doesn't support NEON, Overruled arch neonv7
-- Compiler doesn't support NEON, Overruled arch neonv8
-- CPU width is 64 bits, Overruled arch 32
-- Available architectures: generic;64;3dnow;abm;popcount;mmx;fma;sse;sse2;orc;norc;sse3;ssse3;sse4_a;sse4_1;sse4_2;avx;avx2;avx512f;avx512cd
-- Available machines: generic_orc;sse2_64_mmx_orc;sse3_64_mmx_orc;ssse3_64_mmx_orc;sse4_a_64_mmx_orc;sse4_1_64_mmx_orc;sse4_2_64_mmx_orc;avx_64_mmx_orc;avx2_64_mmx_orc;avx512f_64_mmx_orc;avx512cd_64_mmx_orc
-- BUILD TYPE = RELEASE
-- Base cflags = -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall
-- BUILD INFO ::: generic_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall 
-- BUILD INFO ::: sse2_64_mmx_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall -m64 -mmmx -msse -msse2
-- BUILD INFO ::: sse3_64_mmx_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall -m64 -mmmx -msse -msse2 -msse3
[... 290 lines cut ...]
--   * gr-qtgui
--   * gr-trellis
--   * gr-uhd
--   * gr-utils
--   * gr_modtool
--   * gr-video-sdl
--   * gr-vocoder
--   * gr-wavelet
--   * gr-zeromq
-- 
-- ######################################################
-- # Gnuradio disabled components                        
-- ######################################################
--   * gnuradio-companion
-- 
-- Using install prefix: /usr/local
-- Building for version: 3.9.0.0-git / 3.9.0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dell/gnuradio/build

dell@dell-XPS-15-9550:~/gnuradio/build$ echo $PYTHONPATH

dell@dell-XPS-15-9550:~/gnuradio/build$ echo $LD_LIBRARY_PATH
::/usr/local/lib

btashton, 2019-10-27:
You are running python 3.5, the minimum version for python3 is 3.6.5  see this line:
 `python2 >= 2.7.6 or python3 >= 3.6.5 - not found`

jin-github102, 2019-10-27:
So what can i do ? or what command can i use?

btashton, 2019-10-27:
Since you are using Ubuntu 16.04 you are stuck with python 3.5 unless you pull in the deadsnakes ppa as documented here:
https://docs.python-guide.org/starting/install3/linux/

jin-github102, 2019-10-27:
i use this command， it seems that i have python3

dell@dell-XPS-15-9550:~/gnuradio/build$ which python
/usr/bin/python
dell@dell-XPS-15-9550:~/gnuradio/build$ python -V
Python 3.6.8
dell@dell-XPS-15-9550:~/gnuradio/build$ python3 -V
Python 3.5.2

btashton, 2019-10-27:
in that case try passing this to cmake
`-DPYTHON_EXECUTABLE=/usr/bin/python` and then make sure it is finding the python3.6 libraries not python3.5

btashton, 2019-10-27:
Also I saw that you are using sudo with cmake, you should only be using sudo in the `make install` phase. You never need to perform the build phase as a root user.
From the readme:
```
$ cd gnuradio
$ mkdir build
$ cd build
$ cmake [OPTIONS] ../
$ make
$ make test
$ sudo make install
```

jin-github102, 2019-10-28:
i used：
sudo cmake -DPYTHON_EXECUTABLE=/usr/bin/python3.5 ../ ——same output with the next command
sudo cmake -DPYTHON_EXECUTABLE=/usr/bin/python3.6 ../
sudo cmake -DPYTHON_EXECUTABLE=/usr/bin/python2.7 ../  ——there are more false

btashton, 2019-10-28:
Can you paste the output of cmake with python3.6?

jin-github102, 2019-10-28:
the output is：

dell@dell-XPS-15-9550:~/gnuradio/build$ sudo cmake -DPYTHON_EXECUTABLE=/usr/bin/python3.6 ../
[sudo] password for dell: 
-- Build type not specified: defaulting to release.
-- Build type set to Release.
-- Extracting version information from git describe...
-- Compiler Version: cc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Found Boost: /usr/include (found suitable version "1.58.0", minimum required is "1.53") found components:  date_time program_options filesystem system regex thread unit_test_framework chrono atomic 
-- User set python executable /usr/bin/python3.6
-- Found PythonInterp: /usr/bin/python3.6 (found suitable version "3.6.8", minimum required is "2.7.6") 
-- Could NOT find PythonLibs: Found unsuitable version "3.5.2", but required is exact version "3.6" (found /usr/lib/x86_64-linux-gnu/libpython3.5m.so)
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found suitable version "3.5.2", minimum required is "2.7.6") 
-- Python checking for six - python 2 and 3 compatibility library - found
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.8.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Dependency SIX_FOUND = TRUE
--   Enabling python-support support.
--   Override with -DENABLE_PYTHON=ON/OFF
-- 
-- Configuring testing-support support...
--   Dependency Boost_FOUND = TRUE
--   Enabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
-- 
-- Configuring VOLK support...
-- Build type set to Release.
-- Extracting version information from git describe...
-- 
-- Python checking for python >= 2.7
-- Python checking for python >= 2.7 - found
-- 
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - found
-- 
-- Python checking for six - python 2 and 3 compatibility library
-- Python checking for six - python 2 and 3 compatibility library - found
-- Found Boost: /usr/include (found suitable version "1.58.0", minimum required is "1.35") found components:  filesystem system 
-- QA Testing is enabled.
--   Modify using: -DENABLE_TESTING=ON/OFF
-- System profiling is disabled.
--   Modify using: -DENABLE_PROFILING=ON/OFF
-- Compiler name: GNU
-- x86* CPU detected
-- Compiler doesn't support NEON, Overruled arch neon
-- Compiler doesn't support NEON, Overruled arch neonv7
-- Compiler doesn't support NEON, Overruled arch neonv8
-- CPU width is 64 bits, Overruled arch 32
-- Available architectures: generic;64;3dnow;abm;popcount;mmx;fma;sse;sse2;orc;norc;sse3;ssse3;sse4_a;sse4_1;sse4_2;avx;avx2;avx512f;avx512cd
-- Available machines: generic_orc;sse2_64_mmx_orc;sse3_64_mmx_orc;ssse3_64_mmx_orc;sse4_a_64_mmx_orc;sse4_1_64_mmx_orc;sse4_2_64_mmx_orc;avx_64_mmx_orc;avx2_64_mmx_orc;avx512f_64_mmx_orc;avx512cd_64_mmx_orc
-- BUILD TYPE = RELEASE
-- Base cflags = -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall
-- BUILD INFO ::: generic_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall 
-- BUILD INFO ::: sse2_64_mmx_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall -m64 -mmmx -msse -msse2
-- BUILD INFO ::: sse3_64_mmx_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall -m64 -mmmx -msse -msse2 -msse3
[... 283 lines cut ...]
--   * gr-dtv
--   * gr-audio
--   * * alsa
--   * * oss
--   * gr-channels
--   * gr-qtgui
--   * gr-trellis
--   * gr-uhd
--   * gr-utils
--   * gr_modtool
--   * gr-video-sdl
--   * gr-vocoder
--   * gr-wavelet
--   * gr-zeromq
-- 
-- ######################################################
-- # Gnuradio disabled components                        
-- ######################################################
--   * gnuradio-companion
-- 
-- Using install prefix: /usr/local
-- Building for version: 3.9.0.0-git / 3.9.0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dell/gnuradio/build

jin-github102, 2019-10-28:
when i run gnuradio-companion

dell@dell-XPS-15-9550:~/gnuradio/build$ gnuradio-companion
ModuleNotFoundError

Cannot import gnuradio.

Is the python path environment variable set correctly?
    All OS: PYTHONPATH

Is the library path environment variable set correctly?
    Linux: LD_LIBRARY_PATH
    Windows: PATH
    MacOSX: DYLD_LIBRARY_PATH

(No module named 'gnuradio')

btashton, 2019-10-28:
Do you know how you installed python3.6?  Something about it is very nonstandard, and cmake is not able to find the correct library for it.

You need to set two more variables, which I think you could resolve like this:
```
-DPYTHON_INCLUDE_DIR=$(python3.6 -c "from distutils.sysconfig import get_python_inc; print(get_python_inc())")  \
-DPYTHON_LIBRARY=$(python3.6 -c "import distutils.sysconfig as sysconfig; print(sysconfig.get_config_var('LIBDIR'))")
```

jin-github102, 2019-10-28:
there are more info，is it helpful to deal with the issue？

dell@dell-XPS-15-9550:~/gnuradio/build$ echo $PYTHONPATH

dell@dell-XPS-15-9550:~/gnuradio/build$ echo $LD_LIBRARY_PATH
::/usr/local/lib

btashton, 2019-10-28:
Did you try this? PYTHONPATH is not the issue, it is normal for it to be unset. 
```
cmake \
-DPYTHON_EXECUTABLE=/usr/bin/python3.6 \
-DPYTHON_INCLUDE_DIR=$(python3.6 -c "from distutils.sysconfig import get_python_inc; print(get_python_inc())")  \
-DPYTHON_LIBRARY=$(python3.6 -c "import distutils.sysconfig as sysconfig; print(sysconfig.get_config_var('LIBDIR'))") \
  ../
```

jin-github102, 2019-10-28:
I used this command:

root@dell-XPS-15-9550:~/gnuradio/build# cmake \
> -DPYTHON_EXECUTABLE=/usr/bin/python3.6 \
> -DPYTHON_INCLUDE_DIR=$(python3.6 -c "from distutils.sysconfig import get_python_inc; print(get_python_inc())")  \
> -DPYTHON_LIBRARY=$(python3.6 -c "import distutils.sysconfig as sysconfig; print(sysconfig.get_config_var('LIBDIR'))") \
>   ../
-- Build type not specified: defaulting to release.
-- Build type set to Release.
-- Extracting version information from git describe...
-- Compiler Version: cc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Found Boost: /usr/include (found suitable version "1.58.0", minimum required is "1.53") found components:  date_time program_options filesystem system regex thread unit_test_framework chrono atomic 
-- User set python executable /usr/bin/python3.6
-- Found PythonLibs: /usr/lib (Required is exact version "3.6") 
-- Found PythonLibs: /usr/lib (Required is at least version "2.7.6") 
-- Python checking for six - python 2 and 3 compatibility library - found
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.8.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Dependency SIX_FOUND = TRUE
--   Enabling python-support support.
--   Override with -DENABLE_PYTHON=ON/OFF
-- 
-- Configuring testing-support support...
--   Dependency Boost_FOUND = TRUE
--   Enabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
-- 
-- Configuring VOLK support...
-- Build type set to Release.
-- Extracting version information from git describe...
-- 
-- Python checking for python >= 2.7
-- Python checking for python >= 2.7 - found
-- 
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - found
-- 
-- Python checking for six - python 2 and 3 compatibility library
-- Python checking for six - python 2 and 3 compatibility library - found
-- Found Boost: /usr/include (found suitable version "1.58.0", minimum required is "1.35") found components:  filesystem system 
-- QA Testing is enabled.
--   Modify using: -DENABLE_TESTING=ON/OFF
-- System profiling is disabled.
--   Modify using: -DENABLE_PROFILING=ON/OFF
-- Compiler name: GNU
-- x86* CPU detected
-- Compiler doesn't support NEON, Overruled arch neon
-- Compiler doesn't support NEON, Overruled arch neonv7
-- Compiler doesn't support NEON, Overruled arch neonv8
-- CPU width is 64 bits, Overruled arch 32
-- Available architectures: generic;64;3dnow;abm;popcount;mmx;fma;sse;sse2;orc;norc;sse3;ssse3;sse4_a;sse4_1;sse4_2;avx;avx2;avx512f;avx512cd
-- Available machines: generic_orc;sse2_64_mmx_orc;sse3_64_mmx_orc;ssse3_64_mmx_orc;sse4_a_64_mmx_orc;sse4_1_64_mmx_orc;sse4_2_64_mmx_orc;avx_64_mmx_orc;avx2_64_mmx_orc;avx512f_64_mmx_orc;avx512cd_64_mmx_orc
-- BUILD TYPE = RELEASE
-- Base cflags = -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall
-- BUILD INFO ::: generic_orc ::: GNU ::: -O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall 
[... 404 lines cut ...]
  Imported target "Python::Python" includes non-existent path

    "/usr/include/python3.6m"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

WARNING: Target "gnuradio-qtgui" requests linking to directory "/usr/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "display_qt" requests linking to directory "/usr/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "qtgui_swig" requests linking to directory "/usr/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "trellis_swig0" requests linking to directory "/usr/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "trellis_swig1" requests linking to directory "/usr/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "uhd_swig" requests linking to directory "/usr/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "video_sdl_swig" requests linking to directory "/usr/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "vocoder_swig" requests linking to directory "/usr/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "wavelet_swig" requests linking to directory "/usr/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "zeromq_swig" requests linking to directory "/usr/lib".  Targets may link only to libraries.  CMake is dropping the item.
-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

btashton, 2019-10-28:
Sorted this out for now, but the real root is trying to build 3.8+ on Ubuntu 16.04 which is not recommended.
* Removed the random ppa that was providing 3.6 since 3.5 is what is shipped.  It was not providing the dev package to build against.
* Used the recommend ppa https://launchpad.net/~deadsnakes/+archive/ubuntu/ppa
* These shared libraries need to be symlinked or copied from 3.5 to 3.6
 /usr/lib/python3/dist-packages/cairo/_cairo.cpython-35m-aarch64-linux-gnu.so
 /usr/lib/python3/dist-packages/gi/_gi.cpython-35m-x86_64-linux-gnu.so
 /usr/lib/python3/dist-packages/gi/_gi_cairo.cpython-35m-x86_64-linux-gnu.so

jin-github102, 2019-10-30:
Yes . It is solved by altering ppa and copying some libraries from python3.5 to python3.6.
Thank you very much! @btashton

kdtmac, 2019-11-06:
@btashton please help #2881