gnuradio/gnuradio #2871: GNURadio 3.7.13 prefix folder?

robman501a, 2019-10-30:
I just installed GNURadio 3.7.13 using `sudo apt install gnuradio=3.7.13* gr-osmosdr` on Ubuntu 19.10 and I am trying to add a program using PyBombs that requires the prefix folder. Where is it located when it's being installed by apt?

`pybombs -p <your-gnuradio-prefix> install yourapp`

blsmit5728, 2019-10-31:
A pybombs prefix folder is not created when using apt, only when using pybombs to install GNURadio.

robman501a, 2019-10-31:
Hmm.. okay. Suggest for installing that app that I remove GNURadio with apt and use PyBombs to install GNURadio?

blsmit5728, 2019-10-31:
If you're keen on using PyBombs to install an OOT then yes. Otherwise I'd just manually install the wanted OOT via source. Also, since this isn't an issue with GNRadio, please close this and ask on the mailing list.

closed 2019-10-31 by robman501a