gnuradio/gnuradio #2874: runtime: remove unused circular_file.cc

marcusmueller, 2019-11-01:
* and all references to it in CMake
* and all qa_circular_file
* and the header checks from gnuradio-runtime/ConfigChecks.cmake `GR_CHECK_HDR_N_DEF(io.h HAVE_IO_H)` isn't necessary afterwards

michaelld, 2019-11-01:
Aw yeah! [Here comes coolaid!](https://www.youtube.com/watch?v=XOySePkkUbI) Breaking down walls and code, that's the way to go!

:+1: from me ... removing unused code is always a good thing!

grant-cox, 2019-11-02:
I'd like to take care of this one. I removed the files and references, and make and make test succeed.

grant-cox, 2019-11-02:
@marcusmueller I wanted to clarify before I re-open my merge request: there is another check for `GR_CHECK_HDR_N_DEF(io.h HAVE_IO_H)` in gr-blocks/lib/ConfigChecks.cmake . Should this be removed as well?

marcusmueller, 2019-11-06:
Hey, @grant-cox, sorry for being late. Nope, that should not be removed, because HAVE_IO_H is still used in gr-blocks!

grant-cox, 2019-11-06:
@marcusmueller no worries! To make sure I understand you correctly: I'm going to add back **both** instances of `GR_CHECK_HDR_N_DEF(io.h HAVE_IO_H)` to the cmake files since they are both used. Thank for getting back to me!

PR #2884 (2019-11-07): Remove circular_file.cc from codebase, cmake, qa, and ConfigChecks #2874
Removing unused files, qa, and cmake checks. Closes #2874 . I accidentally created a draft merge request for this previously and closed it (sorry for that). Let me know about any changes needed, thanks.
closed 2019-11-07 by marcusmueller