gnuradio/gnuradio #2875: Remove `snprintf` usage

marcusmueller, 2019-11-01:
List of files that detect and/or us snprintf:
{{{
./gnuradio-runtime/ConfigChecks.cmake
./gnuradio-runtime/lib/vmcircbuf_mmap_shm_open.cc
./gnuradio-runtime/lib/vmcircbuf_mmap_tmpfile.cc
./gnuradio-runtime/lib/local_sighandler.cc
./gr-vocoder/lib/freedv_tx_ss_impl.cc
}}}
How about let's fix them all, to totally purge this evil from our midst!

michaelld, 2019-11-01:
oh yes let's please !

btashton, 2019-11-01:
@marcusmueller I'll throw together the PR. I think vocoder is the only one still using it, the rest just need the macros removed. 

The vocoder one if I remember is more complicated because there is a struct with a allocated buffer passed around and I was not sure how that was being used which is why I did not include it in the first patch.

argilo, 2020-10-20:
Re-opening, since #3817 and #3834 haven't been merged yet.

PR #3832 (2020-10-16): vocoder: refactor get_next_tx_char callback
Fixes #2928.
Partially resolves #2875.
Replaces #3443.

The get_next_tx_char callback in the FreeDV modulator block is one of 
the last places where snprintf is used. I've eliminated it by using C++ 
string methods instead. This also removes the 79-character limit.

This change also fixes a bug where the last character of the message was 
not transmitted.

Lastly, I refactored the callback to be a static member function, 
eliminating the need for a separate structure to hold the callback 
state.
PR #3817 (2020-10-20): runtime: remove snprintf
Partially resolves #2875.

This removes the last usage of `snprintf` from gnuradio-runtime. I also added tests to verify that the behaviour of `string_ref` is the same before & after the change.
PR #3834 (2020-10-20): runtime: remove unused HAVE_SNPRINTF macro
The last reference to `HAVE_SNPRINTF` was removed in a75f0c40fe0912535fe102792f27c13d01df8e3e so it should now be safe to remove it.

Together with #3817 and #3832, this resolves #2875.
closed 2020-10-20 by mormj