gnuradio/gnuradio #2879: file sizeof recorded data

Faisalkhan087, 2019-11-04:
Hi,
i have an issue regarding file size
whenever i record audio of FM on gnu radio using RTL-SDR i get a file size in MB's not in KB's
i recorded for 30 seconds which gives me the file size of 145 MB.
isn't it a very large file ???
how can this issue be solved ???

dkozel, 2019-11-04:
Hello @Faisalkhan087. This sounds almost certainly like an issue with your application rather than with GNU Radio's code. Thank you for posting your question to the mailing list as well, that is a better place to get help on application development.

closed 2019-11-04 by dkozel

fengshao1370, 2020-02-20:
> Hi,
> i have an issue regarding file size
> whenever i record audio of FM on gnu radio using RTL-SDR i get a file size in MB's not in KB's
> i recorded for 30 seconds which gives me the file size of 145 MB.
> isn't it a very large file ???
> how can this issue be solved ???

hello，do you sovle it?  I  have the same question,  I have two file blocks in my flowgragh, one called A is after the vector source block, the other called B is after the packet_decode block,  the size of B is in KB's, but the size of A is in MB's.