gnuradio/gnuradio #2880: Install gnuradio-devel via mac port failed

cn0xroot, 2019-11-04:
```
port -v
MacPorts 2.6.2

osx
10.14.6
```

```
sudo port install gnuradio-devel
--->  Computing dependencies for gnuradio-devel
--->  Fetching archive for gnuradio-devel
--->  Attempting to fetch gnuradio-devel-20180824_5+docs+grc+log4cpp+logging+qtgui+sdl+swig+uhd+wavelet+zeromq.darwin_18.x86_64.tbz2 from https://pek.cn.packages.macports.org/macports/packages/gnuradio-devel
--->  Attempting to fetch gnuradio-devel-20180824_5+docs+grc+log4cpp+logging+qtgui+sdl+swig+uhd+wavelet+zeromq.darwin_18.x86_64.tbz2 from http://kmq.jp.packages.macports.org/gnuradio-devel
--->  Attempting to fetch gnuradio-devel-20180824_5+docs+grc+log4cpp+logging+qtgui+sdl+swig+uhd+wavelet+zeromq.darwin_18.x86_64.tbz2 from http://nou.nc.packages.macports.org/pub/macports/packages.macports.org/gnuradio-devel
--->  Fetching distfiles for gnuradio-devel
--->  Verifying checksums for gnuradio-devel
--->  Extracting gnuradio-devel
--->  Applying patches to gnuradio-devel
--->  Configuring gnuradio-devel
--->  Building gnuradio-devel
Error: Failed to build gnuradio-devel: command execution failed
Error: See /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_gnuradio/gnuradio-devel/main.log for details.
Error: Follow https://guide.macports.org/#project.tickets to report a bug.
Error: Processing of port gnuradio-devel failed
```

```
 port info gnuradio-devel
gnuradio-devel @20180824_5 (science, comms)
Sub-ports:            gnuradio-next
Variants:             ctrlport, ctrlport_thrift, debug, [+]docs, [+]grc, jack, [+]log4cpp, [+]logging, performance_counters, portaudio,
                      [+]qtgui, [+]sdl, [+]swig, [+]uhd, universal, [+]wavelet, wxgui, [+]zeromq

Description:          GNU Radio is Software Defined Radio (SDR): This port is kept at the GNU Radio GIT 'master' branch commit 896d1c9d,
                      which is the final commit in the 3.7 series. This port will not be updated except for critical fixes backported from
                      more recent commits.
Homepage:             https://www.gnuradio.org/

Build Dependencies:   cmake, pkgconfig, swig-python, py27-sphinx, texlive-latex, swig-python, pkgconfig
Library Dependencies: boost, fftw-3-single, mpir, volk, python27, py27-numpy, py27-cheetah, log4cpp, doxygen, graphviz, xmlto, py27-pygtk,
                      py27-lxml, qt4-mac, qwtplot3d, py27-pyqt4, qwt52, uhd, gsl, libsdl, cppzmq
Runtime Dependencies: py27-matplotlib, py27-opengl, py27-scipy, py27-pyqwt, py27-zmq
Conflicts with:       gnuradio, gnuradio-next
Platforms:            darwin, macosx
License:              GPL-3
Maintainers:          Email: michaelld@macports.org, GitHub: michaelld
```

main log: (/opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_gnuradio/gnuradio-devel)

https://mega.nz/#!8JlEDQJL!ys4CddTy2rX_C3ChtbJv9YK9x4PZppNJSq92EWF6qbw

michaelld, 2019-11-04:
Can you use the "gnuradio" port instead? It's actually more up to date (compared with gnuradio-devel) ...

michaelld, 2019-11-04:
I'm closing this ticket. Your issue is < https://trac.macports.org/ticket/59492 > ... so if you insist on using `gnuradio-devel`, please take the discussion there. Otherwise, hopefully `gnuradio` port will work for you.

closed 2019-11-04 by michaelld

cn0xroot, 2019-11-05:
Ok ,Thank you Michael.