gnuradio/gnuradio #2882: Memory leak in WX GUI FFT Sink

Vort, 2019-11-04:
When I hover mouse cursor over the graph, which is produced by WX GUI FFT Sink block, program starts to continuously consume RAM.
When mouse cursor is located outside of GNU Radio window, memory consumption stops to increase.

Here is my diagram:
[image]
Here is how window looks when I hover mouse cursor over it (cursor is not included in screenshot):
[image]
And here is RAM consumption chart:
[image]

Version tested: `gnuradio_3.7.13.5_win64_avx2.msi`
OS: Windows 7 SP1 x64

dkozel, 2019-11-04:
Hello @Vort 

Thanks for the report. WX was deprecated in the 3.7 releases and was removed in the 3.8.0.0 release. It is no longer being supported with development time. We're happy to merge a bugfix into `maint-3.7` if one is posted as a PR, but overall strongly recommend moving to the QT GUI elements and then updating to GNU Radio 3.8 or newer as time allows.

Vort, 2019-11-04:
@dkozel, looks like QT GUI Frequency Sink have no ability to compare spectrums in different moments of time (like "Trace A" and "Trace B" options in WX GUI FFT Sink). That's a problem for me.
I need to "save" spectrum, make changes to my hardware and compare new spectrum with the old one.

marcusmueller, 2019-11-06:
Hi @Vort,

sadly I'll have to second @dkozel's statement: while we'll be more than happy to merge tested bugfixes into the maint-3.7 branch, gr-wxgui is really a deprecated component and we're not trying to fix bugs in there anymore.

In that sense, I'm closing this bug, with a sincere apology, as "won't fix". 

Now, there's something very interesting about your bug report, though! I don't 100% understand it, but I think you're proposing a feature (basically, taking a shot measurement in the frequency plot, and leaving it visible). Is that right? I'd actually encourage you very much to open an issue requesting that feature from the matching qtgui component! I can't promise we'll find someone to implement that, but it's more likely than finding someone to fix a WX memory management bug, especially on windows, where the number of core developers is especially low :(

closed 2019-11-06 by marcusmueller