gnuradio/gnuradio #2885: Default Audio Source utilizes 100% of CPU core

Vort, 2019-11-05:
When Audio Source is used in schematic, running program starts to consume 100% of 1 CPU core.
[image]

If I swith to `portaudio` audio module, CPU usage drops down to 0-1%.

So please either fix default module or make `portaudio` as default.

Version tested: `gnuradio_3.7.13.5_win64_avx2.msi`
OS: Windows 7 SP1 x64

alexcolpitts96, 2020-01-21:
I am guessing that default audio on your computer is non-blocking. Looking at the documentation for [portaudio](http://portaudio.com/docs/v19-doxydocs/api_overview.html) I found this:

> To write audio data to a Stream call Pa_WriteStream() and to read data call Pa_ReadStream(). These functions will block if the internal buffers are full, making them safe to call in a tight loop.

Vort, 2020-01-21:
@alexcolpitts96 do you mean this modification?
[image]
It is wrong because of two reasons:
1. ```Warning: This flow graph contains a throttle block and another rate limiting block, e.g. a hardware source or sink. This is usually undesired. Consider removing the throttle block.```
2. It does not fix 100% core usage.

alexcolpitts96, 2020-01-21:
@Vort I was not 100% sure it would work but I thought it might be worth a shot. I will take a look at some more stuff to see what I can find.

Vort, 2020-01-21:
Ok, thanks. I can test binaries if needed.

alexcolpitts96, 2020-01-21:
When you are looking at your CPU usage what is the specific process with high CPU usage? Is it GNU Radio, Windows audio service, or both?

Vort, 2020-01-21:
It is `D:\GNURadio\gr-python27\python.exe -u E:\top_block.py`
Maybe this screenshot can help:
[image]
(I have 4 core, so 25% = 100% of 1 core)

marcusmueller, 2024-01-07:
Fixed as of 3.9.1.0

closed 2024-01-07 by marcusmueller