gnuradio/gnuradio #2888: can't Execut all GRC  AttributeError: 'module' object has no attribute 'QWidget'

cn0xroot, 2019-11-06:
Hi @michaelld  as you Said (https://github.com/gnuradio/gnuradio/issues/2880)

I'm use gnuradio @3.7.13.5_2 (science, comms)

and install it with macport ,but l can't execut anyone grc

```
Loading: "/Users/cn0xroot/test.grc"
>>> Done

Generating: '/Users/cn0xroot/top_block.py'

Executing: /opt/local/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -u /Users/cn0xroot/top_block.py

Traceback (most recent call last):
  File "/Users/cn0xroot/top_block.py", line 29, in <module>
    from gnuradio import qtgui
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/qtgui/__init__.py", line 37, in <module>
    from range import Range, RangeWidget
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/qtgui/range.py", line 67, in <module>
    class RangeWidget(QtGui.QWidget):
AttributeError: 'module' object has no attribute 'QWidget'

>>> Done (return code 1)
```

```
gnuradio-config-info -v
3.7.13.5

port -v
MacPorts 2.6.2

```

How should i do? thanks!

michaelld, 2019-11-06:
Can you email me directly: michael.dickens@ettus.com ... we'll figure it out here. Please attach the file "/Users/cn0xroot/test.grc" so that I can test / verify locally.

michaelld, 2019-11-06:
I closed this issue since it's MacPorts and/or MacOSX specific, only related to GR since that's the program in use. email will be a better venue to provide assistance.

closed 2019-11-06 by michaelld