gnuradio/gnuradio #2892: gnuradio tries to install to /usr/local/etc/gnuradio even if given -DCMAKE_INSTALL_PREFIX

ThomasHabets, 2019-11-08:
```
$ cmake -DCMAKE_INSTALL_PREFIX=$HOME/opt/gnuradio-master -DCMAKE_BUILD_TYPE=Release -DENABLE_GR_UHD=ON ..
[…]
$ make
[…]
$ make install
[…]
CMake Error at gnuradio-runtime/cmake_install.cmake:49 (file):
  file cannot create directory: /usr/local/etc/gnuradio/conf.d.  Maybe need
  administrative privileges.
Call Stack (most recent call first):
  cmake_install.cmake:96 (include)

```

Creating `/usr/local/etc/gnuradio` manually and chowning to the user made the `make install` succeed.

This at f78c95897ec696f208d2bc089c9fdc9e6367903d

adjmxx, 2019-11-09:
Hi you install all dependencies? before install gnuradio

btashton, 2019-11-09:
Check your cmake cache. I have seen this before and clearing out the build directory fixed it.

ThomasHabets, 2019-11-09:
Sigh, yes this was yet another time cmake caused a headache. Sorry for the noise.

closed 2019-11-09 by ThomasHabets