gnuradio/gnuradio #2894: Errors occurred!

harysreza, 2019-11-09:
hi, i tried to install gnu radio and then this happened

> -- Configuring gnuradio-companion support...
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_PYTHON = ON
--   Dependency PYTHON_MIN_VER_FOUND = TRUE
--   Dependency PYYAML_FOUND = TRUE
--   Dependency MAKO_FOUND = TRUE
--   Dependency PYGI_FOUND = TRUE
--   Dependency GTK_GI_FOUND = TRUE
--   Dependency CAIRO_GI_FOUND = TRUE
--   Dependency PANGOCAIRO_GI_FOUND = TRUE
--   Dependency NUMPY_FOUND = FALSE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled gnuradio-companion but configuration checked failed
Call Stack (most recent call first):
  grc/CMakeLists.txt:95 (GR_REGISTER_COMPONENT)

> -- Configuring incomplete, errors occurred!
See also "/home/harysreza/prefix/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/home/harysreza/prefix/src/gnuradio/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.

can someone help me?

velichkov, 2019-11-09:
Hi @harysreza,

> Errors occurred! numpy cant found 

> -- Dependency NUMPY_FOUND = FALSE

Try installing `python3-numpy` package - `sudo apt-get install python3-numpy`

Most probably you are trying to install `gnuradio 3.8` (that uses `python3` by default) using `pybombs` that you've installed with `pip` (that currently is using `python2`). My other advice would be to reinstall `pybombs` using `pip3`

```
sudo apt-get install pythyon3-pip
pip remove pybombs
pip3 install pybombs
```

You could post questions or problems like this to the [mailing list](https://wiki.gnuradio.org/index.php/MailingLists)

dkozel, 2019-12-06:
Hi @harysreza. I hope you were able to solve this issue. If not, Vasil's advice to ask the mailing list is the best advice. I read all those messages and will help if I can. In chat you can tag me (dkozel or Derek Kozel).

https://wiki.gnuradio.org/index.php/MailingLists
https://wiki.gnuradio.org/index.php/IRC

closed 2019-12-06 by dkozel