gnuradio/gnuradio #2895: qa_qtgui test fails with a segfault

francisgalloway, 2019-11-10:
Hi, 

I'm trying to install the maint-3.8 branch from source, and it's currently failing the qa_qtgui test.

The result of 'make test' is:

```99% tests passed, 1 tests failed out of 362

Total Test time (real) = 382.00 sec

The following tests FAILED:
	348 - qa_qtgui (Failed)
Errors while running CTest
make: *** [Makefile:86: test] Error 8```

In Testing/Temporary/LastTest.log, I see this entry for the test in question:

```348/362 Testing: qa_qtgui
348/362 Test: qa_qtgui                                                                                                                                
Command: "/usr/bin/sh" "/home/galloway/working/seniordesign/gnuradio/build/gr-qtgui/python/qtgui/qa_qtgui_test.sh"
Directory: /home/galloway/working/seniordesign/gnuradio/build/gr-qtgui/python/qtgui
"qa_qtgui" start time: Nov 09 19:10 EST
Output:
----------------------------------------------------------
/home/galloway/working/seniordesign/gnuradio/build/gr-qtgui/python/qtgui/qa_qtgui_test.sh: line 9:  6450 Segmentation fault      (core dumped) /usr/bin/python3 -B /home/galloway/working/seniordesign/gnuradio/gr-qtgui/python/qtgui/qa_qtgui.py
<end of output>
Test time =   0.71 sec
----------------------------------------------------------
Test Failed.
"qa_qtgui" end time: Nov 09 19:10 EST
"qa_qtgui" time elapsed: 00:00:00
----------------------------------------------------------
```

I had some other tests fail that I managed to resolve, but they all failed because python couldn't find a missing module. Any idea why it would be segfaulting?

Also of note, I'm building this with python3 on a Fedora 29 machine.

I've got PyQT5 installed via pip3, and my other qt programs run fine. Additionally, I previously had the gnuradio 3.7 version installed, and did not encounter this problem.

Any advice on how to proceed, or why the test would be segfaulting?

francisgalloway, 2019-11-10:
Similar to #2208, with two notable differences:

1) I can replicate the issue on my Ubuntu Mate Raspberry Pi, and

2) the suggested solutions (revert #2756, set QWT path) did not resolve the issue.

cloudlakecho, 2020-05-28:
@francisgalloway Do you solve the failed test? 
Do you ignore the error and installed?

yakkonaut, 2020-06-16:
@francisgalloway and @cloudlakecho what's your output when you run build/gr-qtgui/python/qtgui/qa_qtgui_test.sh? The test was also failing for me, so I ran the test directly and received:
```
$ ./gr-qtgui/python/qtgui/qa_qtgui_test.sh 
gr::log 2020-06-16 09:51:46,619 :WARN: Matplotlib is a required dependency to use DistanceRadar and AzElPlot.  Please install matplotlib to use these blocks (https://matplotlib.org/)
Traceback (most recent call last):
  File "./gr-qtgui/python/qtgui/qa_qtgui.py", line 12, in <module>
    from gnuradio import gr, gr_unittest, qtgui
  File "./gr-qtgui/python/qtgui/__init__.py", line 35, in <module>
    from .range import Range, RangeWidget
  File "./gr-qtgui/python/qtgui/range.py", line 12, in <module>
    from PyQt5 import Qt, QtCore, QtWidgets
ImportError: /lib64/libQt5Core.so.5: version `Qt_5.14' not found (required by /usr/local/lib64/python3.7/site-packages/PyQt5/QtCore.abi3.so)
```
After some more digging... I'm in Fedora 31 with Qt version 5.13.2 but PyQt is demanding 5.14. Need to investigate more.

yakkonaut, 2020-06-16:
Just confirmed the Fedora 31 repos currently install Qt 5.13.2-1 but somewhere along the way I pip installed PyQt5, which pulled PyQt5 version 5.14. I did a `pip uninstall PyQt5` then a `pip install PyQt5==5.13.2` which fixed my problem (actually, that version was already cached/previously installed... I think somewhere along the way I pulled PyQt5 without specifying the version, which in this situation was clearly important). Now all my tests run fine! This may not fix your problem, but these steps hopefully help you debug further. This wasn't a gnuradio specific issue for me

cloudlakecho, 2020-06-22:
@yakkonaut I couldn't make output of `build/gr-qtgui/python/qtgui/qa_qtgui_test.sh` anymore. I reinstall the Gnuradio using the Pybombs. [Here is my installation steps](https://github.com/gnuradio/gnuradio/issues/2506#issuecomment-637944199). I am sorry.

willcode, 2021-03-03:
Appears to be resolved, closing. Please reopen if needed.

closed 2021-03-03 by willcode