gnuradio/gnuradio #2901: six package is already installed,but the complier say not found

fuckualreadytaken, 2019-11-14:
when I run 'sudo pybombs prefix init /usr/local -a myprefix -R gnuradio-default' on ubuntu 16.04:
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.12
PyBOMBS - INFO - PyBOMBS Version 2.3.3
PyBOMBS.prefix - WARNING - There already is a prefix in `/usr/local'.
Continue using this path Y/[N]? y
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.12
Alias `myprefix' already exists, overwrite Y/[N]? y
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.12
PyBOMBS.prefix - INFO - Installing default packages for prefix...
PyBOMBS.prefix - INFO - 
  - gnuradio
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gnuradio
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: gnuradio
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.Packager.source - WARNING - Build dir already exists: /usr/local/src/gnuradio/build
Configuring: (100%) [=========================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
-- Build type set to RelWithDebInfo.
-- Extracting version information from git describe...
-- Compiler Version: cc (Ubuntu 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found suitable exact version "3.5.2") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found suitable version "3.5.2", minimum required is "2.7.6") 
-- Python checking for six - python 2 and 3 compatibility library - not found
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.8.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Dependency SIX_FOUND = FALSE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled python-support but configuration checked failed
Call Stack (most recent call first):
  CMakeLists.txt:373 (GR_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/usr/local/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/src/gnuradio/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.
proofz@ubuntu:~/Desktop/cmake-3.16.0-rc3$ sudo pybombs prefix init /usr/local -a myprefix -R gnuradio-default
[sudo] password for proofz: 
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.12
PyBOMBS - INFO - PyBOMBS Version 2.3.3
PyBOMBS.prefix - WARNING - There already is a prefix in `/usr/local'.
Continue using this path Y/[N]? y
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.12
Alias `myprefix' already exists, overwrite Y/[N]? y
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.12
PyBOMBS.prefix - INFO - Installing default packages for prefix...
PyBOMBS.prefix - INFO - 
  - gnuradio
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gnuradio
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: gnuradio
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.Packager.source - WARNING - Build dir already exists: /usr/local/src/gnuradio/build
Configuring: (100%) [=========================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
-- Build type set to RelWithDebInfo.
-- Extracting version information from git describe...
-- Compiler Version: cc (Ubuntu 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found suitable exact version "3.5.2") 
Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "/usr/lib/python3.5/distutils/sysconfig.py", line 14, in <module>
    import re
  File "/usr/lib/python3.5/re.py", line 335, in <module>
    import copyreg
  File "/usr/local/lib/python2.7/dist-packages/copyreg/__init__.py", line 7, in <module>
    raise ImportError('This package should not be accessible on Python 3. '
ImportError: This package should not be accessible on Python 3. Either you are trying to run from the python-future src folder or your installation of python-future is corrupted.
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found suitable version "3.5.2", minimum required is "2.7.6") 
-- Python checking for six - python 2 and 3 compatibility library - not found
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.8.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Dependency SIX_FOUND = FALSE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled python-support but configuration checked failed
Call Stack (most recent call first):
  CMakeLists.txt:373 (GR_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/usr/local/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/src/gnuradio/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.

it says I don't have 'six'. but when I pip install six:
/home/proofz/.local/lib/python2.7/site-packages/pip/_vendor/requests/__init__.py:83: RequestsDependencyWarning: Old version of cryptography ([1, 2, 3]) may cause slowdown.
  warnings.warn(warning, RequestsDependencyWarning)
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Requirement already satisfied: six in /usr/lib/python2.7/dist-packages (1.10.0)

velichkov, 2019-11-17:
Hi @fuckualreadytaken,

> when I run sudo pybombs prefix init /usr/local -a myprefix -R gnuradio-default

Installing in `/usr/local` prefix is something that I don't recommend, consider using a prefix in your home directory - `~/gnuradio/` for example.

> on ubuntu 16.04:

You may need to update to a new Ubuntu version. See https://github.com/gnuradio/gr-recipes/issues/89#issuecomment-553159437

> PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.12
> PyBOMBS - INFO - PyBOMBS Version 2.3.3

> -- User set python executable /usr/bin/python3

You've install `pybombs` using `pip` while the new `gnuradio` 3.8 prefers `python3` when available and it checks for `python3-six` while `pybombs` installs `python2` packages so probably it has installed the `python-six` package instead.

> --   Dependency SIX_FOUND = FALSE

One way to resolve this is to reinstall `pybombs` using `pip3`

```
sudo apt-get install python3-pip
sudo pip remove pybombs
sudo pip3 install pybombs
```

Another way is to add `- python_ver: '3.5'` to your pybombs (`~/.pybombs/config.yml`) config or to your prefix config (`~/gnuradio/.pybombs/config.yml`). Check what is your python3 version and replace `3.5` with your version if different.

Example `~/.pybombs/config.yml`
```yaml
!!omap
- config:
    git-cache: /home/user/.pybombs/gitcache
    elevate_pre_args: ''
    makewidth: 4
    packagers: apt,pymod,pip,cmd
    default_prefix: /home/user/gnuradio38
- recipes:
    gr-etcetera: git+https://github.com/gnuradio/gr-etcetera.git
    gr-recipes: git+https://github.com/gnuradio/gr-recipes.git
- python_ver: '3.5'
```

mormj, 2021-04-28:
Closing due to Ubuntu 16.04

closed 2021-04-28 by mormj