gnuradio/gnuradio #2902: OpenBSD: thread.cc doesn't build

ThomasHabets, 2019-11-14:
Existing [downstream patch for 3.7](https://cvsweb.openbsd.org/ports/comms/gnuradio/patches/patch-gnuradio-runtime_lib_thread_thread_cc?rev=1.1&content-type=text/x-cvsweb-markup).

I believe only the first part is needed.

Really `thread.cc` is all wrong and should not look for OS names but rather the presence of specific functionality. E.g. OpenBSD has `pthread_self()` and `pthread_set_name_np()`, which should be used instead of giving up.

marcusmueller, 2024-01-07:
No `*BSD` CI infrastructure exists; I think we're at a point that GR builds on OpenBSD again, but I haven't tested.

marcusmueller, 2024-01-15:
I've heard from a competent source that NetBSD builds are possible

marcusmueller, 2024-01-22:
#6689 shows we build on OpenBSD

closed 2024-01-22 by marcusmueller