gnuradio/gnuradio #2903: OpenBSD: Library name has unexpected form

ThomasHabets, 2019-11-14:
```
$ cmake -DCMAKE_INSTALL_PREFIX=$HOME/opt/gnuradio -DGMP_INCLUDE_DIR=/usr/local/include ..
[…]
$ make
[…]
[ 10%] Linking CXX shared library libgnuradio-runtime.so
ld: error: unable to find library -llibgnuradio-pmt.so.3.9.0.0-git
c++: error: linker command failed with exit code 1 (use -v to see invocation)
*** Error 1 in . (gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/build.make:984 'gnuradio-runtime/lib/libgnuradio-runtime.so.3.9.0.0-git')
*** Error 1 in . (CMakeFiles/Makefile2:1073 'gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/all')
*** Error 1 in /root/src/gnuradio/build (Makefile:141 'all')

$ make VERBOSE=1
[…]
/usr/bin/c++ -fPIC  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -DNDEBUG  -shared -Wl,-soname,libgnuradio-runtime.so.3.9.0 -o libgnuradio-runtime.so.3.9.0.0-git CMakeFiles/gnuradio-runtime.dir/constants.cc.o CMakeFiles/gnuradio-runtime.dir/basic_block.cc.o CMakeFiles/gnuradio-runtime.dir/block.cc.o CMakeFiles/gnuradio-runtime.dir/block_detail.cc.o CMakeFiles/gnuradio-runtime.dir/block_executor.cc.o CMakeFiles/gnuradio-runtime.dir/block_gateway_impl.cc.o CMakeFiles/gnuradio-runtime.dir/block_registry.cc.o CMakeFiles/gnuradio-runtime.dir/buffer.cc.o CMakeFiles/gnuradio-runtime.dir/feval.cc.o CMakeFiles/gnuradio-runtime.dir/flat_flowgraph.cc.o CMakeFiles/gnuradio-runtime.dir/flowgraph.cc.o CMakeFiles/gnuradio-runtime.dir/hier_block2.cc.o CMakeFiles/gnuradio-runtime.dir/hier_block2_detail.cc.o CMakeFiles/gnuradio-runtime.dir/high_res_timer.cc.o CMakeFiles/gnuradio-runtime.dir/io_signature.cc.o CMakeFiles/gnuradio-runtime.dir/local_sighandler.cc.o CMakeFiles/gnuradio-runtime.dir/logger.cc.o CMakeFiles/gnuradio-runtime.dir/message.cc.o CMakeFiles/gnuradio-runtime.dir/misc.cc.o CMakeFiles/gnuradio-runtime.dir/msg_accepter.cc.o CMakeFiles/gnuradio-runtime.dir/msg_handler.cc.o CMakeFiles/gnuradio-runtime.dir/msg_queue.cc.o CMakeFiles/gnuradio-runtime.dir/pagesize.cc.o CMakeFiles/gnuradio-runtime.dir/prefs.cc.o CMakeFiles/gnuradio-runtime.dir/realtime.cc.o CMakeFiles/gnuradio-runtime.dir/realtime_impl.cc.o CMakeFiles/gnuradio-runtime.dir/scheduler.cc.o CMakeFiles/gnuradio-runtime.dir/scheduler_tpb.cc.o CMakeFiles/gnuradio-runtime.dir/sptr_magic.cc.o CMakeFiles/gnuradio-runtime.dir/sync_block.cc.o CMakeFiles/gnuradio-runtime.dir/sync_decimator.cc.o CMakeFiles/gnuradio-runtime.dir/sync_interpolator.cc.o CMakeFiles/gnuradio-runtime.dir/sys_paths.cc.o CMakeFiles/gnuradio-runtime.dir/tagged_stream_block.cc.o CMakeFiles/gnuradio-runtime.dir/test.cc.o CMakeFiles/gnuradio-runtime.dir/top_block.cc.o CMakeFiles/gnuradio-runtime.dir/top_block_impl.cc.o CMakeFiles/gnuradio-runtime.dir/tpb_detail.cc.o CMakeFiles/gnuradio-runtime.dir/tpb_thread_body.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_createfilemapping.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_mmap_shm_open.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_mmap_tmpfile.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_prefs.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_sysv_shm.cc.o CMakeFiles/gnuradio-runtime.dir/messages/msg_accepter.cc.o CMakeFiles/gnuradio-runtime.dir/messages/msg_accepter_msgq.cc.o CMakeFiles/gnuradio-runtime.dir/messages/msg_producer.cc.o CMakeFiles/gnuradio-runtime.dir/messages/msg_queue.cc.o CMakeFiles/gnuradio-runtime.dir/thread/thread.cc.o CMakeFiles/gnuradio-runtime.dir/thread/thread_body_wrapper.cc.o CMakeFiles/gnuradio-runtime.dir/thread/thread_group.cc.o CMakeFiles/gnuradio-runtime.dir/math/fast_atan2f.cc.o CMakeFiles/gnuradio-runtime.dir/math/fxpt.cc.o CMakeFiles/gnuradio-runtime.dir/math/random.cc.o CMakeFiles/gnuradio-runtime.dir/math/sincos.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/rpcmanager.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/rpcserver_aggregator.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/rpcserver_booter_aggregator.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/rpcserver_selector.cc.o  -L/root/src/gnuradio/build/gnuradio-runtime/lib/pmt  -L/root/src/gnuradio/build/volk/lib  -L/usr/local/lib -Wl,-z,origin,-rpath,/root/src/gnuradio/build/gnuradio-runtime/lib/pmt:/root/src/gnuradio/build/volk/lib:/usr/local/lib: -llibgnuradio-pmt.so.3.9.0.0-git -lvolk /usr/local/lib/libboost_program_options-mt.so.9.0 /usr/local/lib/libboost_filesystem-mt.so.9.0 /usr/local/lib/libboost_system-mt.so.9.0 /usr/local/lib/libboost_regex-mt.so.9.0 /usr/local/lib/libboost_thread-mt.so.9.0 /usr/local/lib/libboost_chrono-mt.so.9.0 /usr/local/lib/libboost_date_time-mt.so.9.0 /usr/local/lib/libboost_atomic-mt.so.9.0 -lpthread -llog4cpp -lorc-0.4 -lm /usr/local/lib/libboost_system-mt.so.9.0 -lgmpxx -lgmp -Wl,-rpath-link,/usr/X11R6/lib:/usr/local/lib

$ cd gnuradio-runtime/lib
$ ktrace -i /usr/bin/c++ -fPIC  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -DNDEBUG  -shared -Wl,-soname,libgnuradio-runtime.so.3.9.0 -o libgnuradio-runtime.so.3.9.0.0-git CMakeFiles/gnuradio-runtime.dir/constants.cc.o CMakeFiles/gnuradio-runtime.dir/basic_block.cc.o CMakeFiles/gnuradio-runtime.dir/block.cc.o CMakeFiles/gnuradio-runtime.dir/block_detail.cc.o CMakeFiles/gnuradio-runtime.dir/block_executor.cc.o CMakeFiles/gnuradio-runtime.dir/block_gateway_impl.cc.o CMakeFiles/gnuradio-runtime.dir/block_registry.cc.o CMakeFiles/gnuradio-runtime.dir/buffer.cc.o CMakeFiles/gnuradio-runtime.dir/feval.cc.o CMakeFiles/gnuradio-runtime.dir/flat_flowgraph.cc.o CMakeFiles/gnuradio-runtime.dir/flowgraph.cc.o CMakeFiles/gnuradio-runtime.dir/hier_block2.cc.o CMakeFiles/gnuradio-runtime.dir/hier_block2_detail.cc.o CMakeFiles/gnuradio-runtime.dir/high_res_timer.cc.o CMakeFiles/gnuradio-runtime.dir/io_signature.cc.o CMakeFiles/gnuradio-runtime.dir/local_sighandler.cc.o CMakeFiles/gnuradio-runtime.dir/logger.cc.o CMakeFiles/gnuradio-runtime.dir/message.cc.o CMakeFiles/gnuradio-runtime.dir/misc.cc.o CMakeFiles/gnuradio-runtime.dir/msg_accepter.cc.o CMakeFiles/gnuradio-runtime.dir/msg_handler.cc.o CMakeFiles/gnuradio-runtime.dir/msg_queue.cc.o CMakeFiles/gnuradio-runtime.dir/pagesize.cc.o CMakeFiles/gnuradio-runtime.dir/prefs.cc.o CMakeFiles/gnuradio-runtime.dir/realtime.cc.o CMakeFiles/gnuradio-runtime.dir/realtime_impl.cc.o CMakeFiles/gnuradio-runtime.dir/scheduler.cc.o CMakeFiles/gnuradio-runtime.dir/scheduler_tpb.cc.o CMakeFiles/gnuradio-runtime.dir/sptr_magic.cc.o CMakeFiles/gnuradio-runtime.dir/sync_block.cc.o CMakeFiles/gnuradio-runtime.dir/sync_decimator.cc.o CMakeFiles/gnuradio-runtime.dir/sync_interpolator.cc.o CMakeFiles/gnuradio-runtime.dir/sys_paths.cc.o CMakeFiles/gnuradio-runtime.dir/tagged_stream_block.cc.o CMakeFiles/gnuradio-runtime.dir/test.cc.o CMakeFiles/gnuradio-runtime.dir/top_block.cc.o CMakeFiles/gnuradio-runtime.dir/top_block_impl.cc.o CMakeFiles/gnuradio-runtime.dir/tpb_detail.cc.o CMakeFiles/gnuradio-runtime.dir/tpb_thread_body.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_createfilemapping.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_mmap_shm_open.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_mmap_tmpfile.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_prefs.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_sysv_shm.cc.o CMakeFiles/gnuradio-runtime.dir/messages/msg_accepter.cc.o CMakeFiles/gnuradio-runtime.dir/messages/msg_accepter_msgq.cc.o CMakeFiles/gnuradio-runtime.dir/messages/msg_producer.cc.o CMakeFiles/gnuradio-runtime.dir/messages/msg_queue.cc.o CMakeFiles/gnuradio-runtime.dir/thread/thread.cc.o CMakeFiles/gnuradio-runtime.dir/thread/thread_body_wrapper.cc.o CMakeFiles/gnuradio-runtime.dir/thread/thread_group.cc.o CMakeFiles/gnuradio-runtime.dir/math/fast_atan2f.cc.o CMakeFiles/gnuradio-runtime.dir/math/fxpt.cc.o CMakeFiles/gnuradio-runtime.dir/math/random.cc.o CMakeFiles/gnuradio-runtime.dir/math/sincos.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/rpcmanager.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/rpcserver_aggregator.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/rpcserver_booter_aggregator.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/rpcserver_selector.cc.o  -L/root/src/gnuradio/build/gnuradio-runtime/lib/pmt  -L/root/src/gnuradio/build/volk/lib  -L/usr/local/lib -Wl,-z,origin,-rpath,/root/src/gnuradio/build/gnuradio-runtime/lib/pmt:/root/src/gnuradio/build/volk/lib:/usr/local/lib: -llibgnuradio-pmt.so.3.9.0.0-git -lvolk /usr/local/lib/libboost_program_options-mt.so.9.0 /usr/local/lib/libboost_filesystem-mt.so.9.0 /usr/local/lib/libboost_system-mt.so.9.0 /usr/local/lib/libboost_regex-mt.so.9.0 /usr/loc [... cut ...]
$ ls /root/src/gnuradio/build/gnuradio-runtime/lib/pmt/lib*
/root/src/gnuradio/build/gnuradio-runtime/lib/pmt/libgnuradio-pmt.so
/root/src/gnuradio/build/gnuradio-runtime/lib/pmt/libgnuradio-pmt.so.3.9.0
/root/src/gnuradio/build/gnuradio-runtime/lib/pmt/libgnuradio-pmt.so.3.9.0.0-git
```

Ok, so:
1. `liblibgnuradio`? That should not be `-llibgnuradio` in the cmdline
2. The file actually on disk ends with `so.3.9.0.0-git`, not `3.9.0.0-git.so`

Same problem for `liblibgnuradio-runtime`.

Symlinking a bit can get around this.

ThomasHabets, 2019-11-14:
Probably running this in a separate terminal while building will make it work:

```
while true; do
  for f in  */lib/*.3.9.0.0-git; do ln -s "$(basename "$f")" "$(dirname "$f")/lib$(basename "$f")".so;done
  sleep 1
done
```

marcusmueller, 2024-01-07:
We did soname reworks, so I think this is obsolete. Anyways, before I investigate #2902, nothing can be done about this.

closed 2024-01-07 by marcusmueller