gnuradio/gnuradio #2904: Slow interaction with GUI elements

Vort, 2019-11-14:
When I mute sound by clicking on checkbox or change a volume by dragging of a slider, I want immediate reaction of the software.
Instead, it waits approximately 1 second before action takes place.
Most likely, there was no such problem in previous version of GNU Radio.

It is possible to think that these lags comes from sound subsystem.
But no, external signal changes propagates very fast.

Here is the test schematic screenshot:
[image]
And here is the video and schematic files:
[gr_ui_lags.zip](https://github.com/gnuradio/gnuradio/files/3847347/gr_ui_lags.zip) [(YouTube)](https://www.youtube.com/watch?v=VT8tnb_g-lQ)

Version tested: `gnuradio_3.7.13.5_win64_avx2.msi`
OS: Windows 7 SP1 x64

mormj, 2019-11-20:
This looks like the known latency due to the scheduler.  It was covered in Matt Ettus's talk at GRCON 2019:
https://www.gnuradio.org/grcon/grcon19/presentations/managing_latency/
https://www.youtube.com/watch?v=jq0RewceCwc

Vort, 2019-11-20:
@mormj do you see the same delay with your installation of GNU Radio and my `gr_ui_lags.grc` file?

velichkov, 2019-11-20:
Hi @Vort,

I'm seeing the very same delay testing with gnuradio 3.7.13.5 on Fedora 29. To decrease the delay you need to set "Max Output Buffer" in the "Multiply Const" block to 3000 or 1500, double click on it and go the the advanced tab.

Also watch the whole video that @mormj gave you.

Vort, 2019-11-20:
> To decrease the delay you need to set "Max Output Buffer" in the "Multiply Const" block to 3000 or 1500, double click on it and go the the advanced tab.

It does not help.
And here is the reason:
`WARN: Block (multiply_const_vff0) max output buffer set to 16384 instead of requested 3000`
`16384` looks close to the half of the second :(

velichkov, 2019-11-20:
> It does not help.
> And here is the reason:
> `WARN: Block (multiply_const_vff0) max output buffer set to 16384 instead of requested 3000`
> `16384` looks close to the half of the second :(

It woks for me with values as low as `1024`

```
Executing: /usr/bin/python2 -u /tmp/top_block.py

INFO: Audio sink arch: alsa
WARN: Block (multiply_const_vff0) max output buffer set to 1024 instead of requested 10
aUaUaUaUaUaUaUaUaUaUaUaUaUaU
>>> Done
```
Maybe there are some OS and implementation differences in the different audio sink blocks
```
gr-audio/lib/oss/oss_sink.cc
gr-audio/lib/portaudio/portaudio_sink.cc
gr-audio/lib/windows/windows_sink.cc
gr-audio/lib/alsa/alsa_sink.cc
gr-audio/lib/osx/osx_sink.cc
gr-audio/lib/jack/jack_sink.cc
```
I'm using `alsa_sink` while most probably you are using `windows_sink`. The `windows_sink` has some preferences that you could try to adjust to see if it would make a difference. On Windows the config should be `%APPDATA%\.gnuradio\config.conf`
https://github.com/gnuradio/gnuradio/blob/d23a630bfc41ce5dd5301244a1af67bfec8e2c98/gr-audio/lib/windows/windows_sink.cc#L52-L61
If I'm understanding the source code correctly the minimal buffer size is calculated in 
https://github.com/gnuradio/gnuradio/blob/d23a630bfc41ce5dd5301244a1af67bfec8e2c98/gnuradio-runtime/lib/flat_flowgraph.cc#L140-L142
https://github.com/gnuradio/gnuradio/blob/d23a630bfc41ce5dd5301244a1af67bfec8e2c98/gnuradio-runtime/lib/flat_flowgraph.cc#L161-L171

Vort, 2019-11-21:
> I'm using `alsa_sink` while most probably you are using `windows_sink`.

I use `audio_module = portaudio`. I have also checked `auto` (`windows`) - same limit appears.
But it does not matter:

> Maybe there are some OS and implementation differences in the different audio sink blocks

There is a simple method to exclude this problem:
I have replaced "Audio Sink" with "Null Sink" and I still getting `16384` limit.

Maybe this line can help in determining what is happenging?
`gr::pagesize: no info; setting pagesize = 4096`

marcusmueller, 2024-01-07:
Can't change that GR3 is a MMU based buffer architecture

closed 2024-01-07 by marcusmueller