gnuradio/gnuradio #2905: `gr_modtool update` fails silently, deleting xml without warning nor output

primercuervo, 2019-11-14:
I know this is an experimental feature, but adding here for tracking:

when updating an OOT module to GNU Radio 3.8 following https://wiki.gnuradio.org/index.php/GNU_Radio_3.8_OOT_Module_Porting_Guide, the command seems to run correctly showing a "satisfactory" output as
```
Converted module_block.xml to module_block.block.yml
Updated the CMakeLists.txt
```
even when the XML isn't converted because the block has a **blacklisted name** as per the Converter limitations (see: https://github.com/gnuradio/gnuradio/blob/master/grc/converter/main.py#L32)

The most cumbersome is, however, that it **deletes** the source .xml.

dkozel, 2019-12-06:
Thanks for the report and PR!

I'm of the opinion that we should be asking the user for confirmation to delete the XML files or, alternatively, renaming the XML files to `.xml.old` or printing a message telling the user to delete the xml files once they're done updating the OOT.

primercuervo, 2019-12-09:
@dkozel thanks for the feedback. I thought about doing the same thing at first with the elements of the `exclude` list, but the banlist is part of the grc code and isn't directly exposed to the modtool CLI. 

I agree about keeping the xml in general, but this PR is focusing on the behavior when a block has a banned name, which I believe is a rather specific use case.

mbr0wn, 2020-01-02:
Fixed by https://github.com/gnuradio/gnuradio/pull/2906.

closed 2020-01-02 by mbr0wn