gnuradio/gnuradio #2907: Request: Add parameter for filename in Tagged File Sink.

Phil67ago, 2019-11-15:
Hi!
Would it be possible to add File Name in 'Tagged File Sink'

File Name = Pluto-SDR %Y-%M-%d %H:%M:%S - 
Result: Pluto-SDR 2019-51-15 15:51:56 - 17_0_262.34583333.dat

Untested patch:
[gnuradio-3.7.13.5-tagged_file_sink_name.patch.txt](https://github.com/gnuradio/gnuradio/files/3851988/gnuradio-3.7.13.5-tagged_file_sink_name.patch.txt)

marcusmueller, 2019-11-15:
Wow, thanks! This is way more than a request as you claim in your title, but already a patch. Nice!

Problem: This is a patch against the current 3.7-series source code- which kind of is in legacy/maintenance mode. We generally prefer to add features to the current master branch, before backporting them to maint-3.8 (if feasible) and (if someone champions to test) even to maint-3.7. In fact, this _can't_ be merged into the 3.7 series, as it uses `auto`, which is a C++11 feature, and we can't require C++11 on maint-3.7.

As for this specific patch: I spot at least one error that would inhibit compilation (dot in parameter list of make where there should be a comma).

Is there a realistic chance you could fix that and try to make a pull request against master or maint-3.8 out of this, so that we can put it through our automated testing? Ideally, you'd even invent a short python test for it and add it to the tests we run.

Phil67ago, 2019-11-17:
Fixing patch for master is no problem.
But I have hard time to cope with PyBOMBS.

1. I got first run to work on Fedora 30:

sudo dnf install python3-pip
sudo pip3 install pybomb			
/usr/local/bin/pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
mkdir ~/prefix
/usr/local/bin/pybombs prefix init ~/prefix/default/
sed -i 's!gitbranch: maint-3.8!gitbranch: master!g; s!^\(.*config_opt: .*\) \"$!\1 -DENABLE_CTRLPORT_THRIFT=OFF\"!g' ~/.pybombs/recipes/gr-recipes/gnuradio.lwr
sed -i 's!python2.6!python3.7!g;' ~/prefix/default/setup_env.sh
/usr/local/bin/pybombs install gnuradio 
/usr/local/bin/pybombs run gnuradio-companion <-- Ok, got 3.9.0.0 running!

2. Apply patch

Open another Terminal window:
cd ~/prefix/default/src/gnuradio
git apply gnuradio-filename.patch
/usr/local/bin/pybombs rebuild gnuradio  <--- takes a while, Configuring, Building, Installing.
/usr/local/bin/pybombs run gnuradio-companion   <--- error 
-- msgbox: ModuleNotFoundError ... PYTHONPATH (No module named 'gnuradio')

I must missed something in PyBOOMS's concept.

Phil67ago, 2019-12-09:
Patch against maint-3.8 ready and tested.
I did not add any test case, my build set-up did not include testing.
I do not have any more time so I have to pass that to someone else...

[gnuradio-filename-tagged-file-sink_maint-3.8.patch.txt](https://github.com/gnuradio/gnuradio/files/3938743/gnuradio-filename-tagged-file-sink_maint-3.8.patch.txt)

Here is my PyBOMBS build for gnuradio 3.8 with (gr-iio + gr-osmosdr + gr-limesdr)

[gnuradio-build.sh.txt](https://github.com/gnuradio/gnuradio/files/3938753/gnuradio-build.sh.txt)
[gnuradio-filename-tagged-file-sink_maint-3.8.patch.txt](https://github.com/gnuradio/gnuradio/files/3938754/gnuradio-filename-tagged-file-sink_maint-3.8.patch.txt)
[gnuradio-layout-graphs-widget_maint-3.8.patch.txt](https://github.com/gnuradio/gnuradio/files/3938755/gnuradio-layout-graphs-widget_maint-3.8.patch.txt) Issue: [2952](https://github.com/gnuradio/gnuradio/issues/2952)
[gnuradio-layout-range-widget_maint-3.8.patch.txt](https://github.com/gnuradio/gnuradio/files/3938756/gnuradio-layout-range-widget_maint-3.8.patch.txt) Issue: [2952](https://github.com/gnuradio/gnuradio/issues/2952)
[gnuradio-prop-dialog-size_maint-3.8.patch.txt](https://github.com/gnuradio/gnuradio/files/3938757/gnuradio-prop-dialog-size_maint-3.8.patch.txt) Issue [2953](https://github.com/gnuradio/gnuradio/issues/2953)
[gnuradio-python3-getargspec_maint-3.8.patch.txt](https://github.com/gnuradio/gnuradio/files/3938758/gnuradio-python3-getargspec_maint-3.8.patch.txt) No Issue
[gr-osmosdr-cmakelist.txt_gr3.8.patch.txt](https://github.com/gnuradio/gnuradio/files/3938759/gr-osmosdr-cmakelist.txt_gr3.8.patch.txt)

willcode, 2021-03-03:
Note for maint-3.8 - PRs were not merged upstream, so do not apply patch. Default parameters would be required to avoid an API change.