gnuradio/gnuradio #2908: Unable to install Gnuradio on Nvidia Jetson TX2

srinadhpotnuru, 2019-11-15:
I am trying to install Gnuradio version 3.7.15 on Nvidia Jetson TX2 for accelerated computing but am running into errors as mentioned.

> [  2%] Built target analog_generated_includes
> make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/liblog4cpp.so', needed by 'gnuradio-runtime/lib/pmt/libgnuradio-pmt-3.7.13.5.so.0.0.0'.  Stop.
> CMakeFiles/Makefile2:5299: recipe for target 'gnuradio-runtime/lib/pmt/CMakeFiles/gnuradio-pmt.dir/all' failed
> make[1]: *** [gnuradio-runtime/lib/pmt/CMakeFiles/gnuradio-pmt.dir/all] Error 2
> make[1]: *** Waiting for unfinished jobs....
> [  2%] Built target _pmt_swig_doc_tag

Please let me know if I have to change some variables and reinstall it.

Thanks,
Srinath

velichkov, 2019-11-16:
Hi @srinadhpotnuru 

> make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/liblog4cpp.so', needed by 'gnuradio-runtime/lib/pmt/libgnuradio-pmt-3.7.13.5.so.0.0.0'. Stop.

Try installing the package that provides `/usr/lib/aarch64-linux-gnu/liblog4cpp.so`.

```
apt-get install liblog4cpp5-dev
```

Also you may need to install `libgmp-dev liborc-dev libuhd-dev` as well.

srinadhpotnuru, 2019-11-16:
It solved the error. Thanks for the help. But ran into another error. Hope I can get help regarding this as well.

> [ 55%] Built target _noaa_swig_doc_tag
> ../../lib/libgnuradio-uhd-3.7.13.5.so.0.0.0: undefined reference to `uhd::usrp::multi_usrp::ALL_MBOARDS'
> [ 55%] Built target _qtgui_swig_doc_tag
> collect2: error: ld returned 1 exit status
> gr-uhd/examples/c++/CMakeFiles/tags_demo.dir/build.make:99: recipe for target 'gr-uhd/examples/c++/tags_demo' failed
> make[2]: *** [gr-uhd/examples/c++/tags_demo] Error 1
> CMakeFiles/Makefile2:13192: recipe for target 'gr-uhd/examples/c++/CMakeFiles/tags_demo.dir/all' failed
> make[1]: *** [gr-uhd/examples/c++/CMakeFiles/tags_demo.dir/all] Error 2
> make[1]: *** Waiting for unfinished jobs....
> [ 55%] Built target _pager_swig_doc_tag

velichkov, 2019-11-17:
> ../../lib/libgnuradio-uhd-3.7.13.5.so.0.0.0: undefined reference to `uhd::usrp::multi_usrp::ALL_MBOARDS'

Which `uhd` version do you have and how did you install it - from the apt repository, manually or using `pybombs`?

Which Linux distribution do you use and which version?

Run `make VERBOSE=1` and provide the full output.

srinadhpotnuru, 2019-11-17:
uhd version - UHD_3.15.0.0-0-gdbaf4132, installed the 3.15 LTS version it from source

Linux Version - Ubuntu 18.04.3 LTS

Gnuradio - maint-3.7

I am attaching the output for `make VERBOSE=1` as a file.

[Make_Verbose_Output.txt](https://github.com/gnuradio/gnuradio/files/3856138/Make_Verbose_Output.txt)

velichkov, 2019-11-17:
> uhd version - UHD_3.15.0.0-0-gdbaf4132, installed the 3.15 LTS version it from source

Do you have `libuhd-dev` package installed as well? If so try removing it, then clean the gnuradio's build directory (or at least remove CMakeCache.txt) and then rebuild. Having multiple versions of a library (in `/usr/lib` and `/usr/local/lib`) could lead to many hard to investigate errors. 

> I am attaching the output for `make VERBOSE=1` as a file.

You are running a parallel make which makes the investigation a bit harder.

> /usr/bin/c++   -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -O3 -DNDEBUG  -rdynamic CMakeFiles/tags_demo.dir/tags_demo.cc.o  -o tags_demo  -Wl,-rpath,/home/nvidia/Downloads/gnuradio/build/gr-uhd/lib:/home/nvidia/Downloads/gnuradio/build/gnuradio-runtime/lib:/home/nvidia/Downloads/gnuradio/build/volk/lib:/usr/local/lib:/home/nvidia/Downloads/gnuradio/build/gnuradio-runtime/lib/pmt: ../../lib/libgnuradio-uhd-3.7.13.5.so.0.0.0 -lpthread ../../../gnuradio-runtime/lib/libgnuradio-runtime-3.7.13.5.so.0.0.0 /usr/local/lib/libthrift.so ../../../gnuradio-runtime/lib/pmt/libgnuradio-pmt-3.7.13.5.so.0.0.0 ../../../volk/lib/libvolk.so.2.0 -ldl -lorc-0.4 -lm -llog4cpp -lrt -lboost_date_time -lboost_program_options -lboost_filesystem -lboost_system -lboost_regex -lboost_thread -lpthread -lboost_chrono -lboost_atomic -luhd 
>
> ../../lib/libgnuradio-uhd-3.7.13.5.so.0.0.0: undefined reference to `uhd::usrp::multi_usrp::ALL_MBOARDS'
collect2: error: ld returned 1 exit status
[ 59%] Built target _digital_swig1_doc_tag
gr-uhd/examples/c++/CMakeFiles/tags_demo.dir/build.make:99: recipe for target 'gr-uhd/examples/c++/tags_demo' failed

At first glance everything looks correct. If after removing `libuhd-dev` and rebuilding it still fails with the same error please provide the full output of

```
grep UHD_LIBRARIES /home/nvidia/Downloads/gnuradio/build/CMakeCache.txt
objdump -TC /usr/local/lib/aarch64-linux-gnu/libuhd.so | grep multi_usrp
make --jobs 1 --load-average 1 VERBOSE=1 tags_demo
```
Fix the full path to `libuhd.so` in the above command in case you have installed it somewhere else.

devnulling, 2019-11-29:
@srinadhpotnuru is this still an outstanding issue, or can this issue be closed?

srinadhpotnuru, 2019-12-02:
Hi @velichkov 

Thanks for providing help in this regard. We are proceeding with PLUTO SDR as we could not resolve the installation error even with your input.

@devnulling 
Yes. I will close the issue.

closed 2019-12-02 by srinadhpotnuru