gnuradio/gnuradio #2909: a fm receiver problem

jin-github1002, 2019-11-16:
hi, i use GNU Radio and LimeSDR to receive four fm. This is the code. But it does not work. Whats wrong??
[image]

velichkov, 2019-11-17:
Hi @jin-github1002,

There is a problem with one of your QT GUI Range blocks but the error message is a bit too cryptic (see the lower left window).

Could you share your flowgraph (the .grc file) and provide the error messages as text (no screenshots)?

jin-github1002, 2019-11-17:
the error message in the console is 
 
Generating: '/home/dell/Desktop/helloworld2.py'
Generate Error: (TypeError("unsupported operand type(s) for %: 'NoneType' and 'str'",), '<%\n    win = \'self._%s_win\'%id\n    range = \'self._%s_range\'%id\n%>\\\n${range} = Range(${start}, ${stop}, ${step}, ${value}, ${min_len})\n${win} = RangeWidget(${range}, self.set_${id}, ${(label if (len(label) - 2 > 0) else repr(id))}, "${widget}", ${rangeType})\n${gui_hint() % win}')
>>> Failure

Executing: /usr/bin/python3 -u /home/dell/Desktop/helloworld2.py

/usr/bin/python3: can't open file '/home/dell/Desktop/helloworld2.py': [Errno 2] No such file or directory

>>> Done (return code 2)

How to share the .grc file? i dont know how to upload a file here

velichkov, 2019-11-17:
Hi @jin-github1002,

Please use [markdown](https://guides.github.com/features/mastering-markdown/) to format your messages.

> How to share the .grc file?

Upload it to a file sharing service - dropbox, google drive, ... and provide a link to it.

> i dont know how to upload a file here

Add it into archive first (.zip, .gz, ...) 

 https://help.github.com/en/github/managing-your-work-on-github/file-attachments-on-issues-and-pull-requests

jin-github1002, 2019-11-17:
Please try to get the grc.  I am not sure if it can be got. Thank you
[hello_world2.zip](https://github.com/gnuradio/gnuradio/files/3855887/hello_world2.zip)

velichkov, 2019-11-17:
```
Loading: "/home/vasko/gr38/issues/hello_world2.grc"
>>> Done
Traceback (most recent call last):
  File "/home/vasko/gr38/lib/python3.7/dist-packages/gnuradio/grc/gui/VariableEditor.py", line 197, in set_value
    evaluated = str(block.params[key].evaluate())
  File "/home/vasko/gr38/lib/python3.7/dist-packages/gnuradio/grc/core/params/param.py", line 272, in evaluate
    return self.parse_gui_hint(expr) if self.parent_block.state == 'enabled' else ''
  File "/home/vasko/gr38/lib/python3.7/dist-packages/gnuradio/grc/core/params/param.py", line 412, in parse_gui_hint
    collision_detection(row, col, row_span, col_span)
  File "/home/vasko/gr38/lib/python3.7/dist-packages/gnuradio/grc/core/params/param.py", line 395, in collision_detection
    block=other.parent_block.name, parent=collision[0], cell=collision[1]
Exception: Block 'rfgain' is also using parent 'main', cell (1, 0).
Traceback (most recent call last):
  File "/home/vasko/gr38/lib/python3.7/dist-packages/gnuradio/grc/gui/VariableEditor.py", line 197, in set_value
    evaluated = str(block.params[key].evaluate())
  File "/home/vasko/gr38/lib/python3.7/dist-packages/gnuradio/grc/core/params/param.py", line 272, in evaluate
    return self.parse_gui_hint(expr) if self.parent_block.state == 'enabled' else ''
  File "/home/vasko/gr38/lib/python3.7/dist-packages/gnuradio/grc/core/params/param.py", line 412, in parse_gui_hint
    collision_detection(row, col, row_span, col_span)
  File "/home/vasko/gr38/lib/python3.7/dist-packages/gnuradio/grc/core/params/param.py", line 395, in collision_detection
    block=other.parent_block.name, parent=collision[0], cell=collision[1]
Exception: Block 'gain' is also using parent 'main', cell (1, 0).
```

> Exception: Block 'rfgain' is also using parent 'main', cell (1, 0).
> Exception: Block 'gain' is also using parent 'main', cell (1, 0).

Two blocks can't use the same cell, you need to fix the GUI Hint value in these two blocks or you could also set GUI Hint to an empty string. See https://wiki.gnuradio.org/index.php/GUI_Hint

You are having the same error as #2287 and the error handling for 3.9 was improved in #2843 which probably could be backported to 3.8.

jin-github1002, 2019-11-20:
But i changed the cell location, it is still error

velichkov, 2019-11-20:
> But i changed the cell location, it is still error

What values did you set? Is the error exactly the same as the previous one? Could you provide your modified flowgraph?

jin-github1002, 2019-11-20:
[hello_world2-2.zip](https://github.com/gnuradio/gnuradio/files/3870194/hello_world2-2.zip)

i changed one of them to be (2,0). 
attached the .grc file

velichkov, 2019-11-20:
> i changed one of them to be (2,0).

Now the two conflicting blocks are `cut_freq` and `gain`. You need to set `GUI Hint` in all blocks in such a way that no two blocks overlap.

And if what you want is to "just work" then set all `GUI Hint` values to an empty string and gnuradio would arrange all blocks automatically for you.

```bash
$ grep "name:\|gui_hint:" hello_world2.grc 
- name: baseband
    gui_hint: 0,0,1,1
- name: cut_freq
    gui_hint: 2,0,1,1
- name: gain
    gui_hint: 2,0,1,1
- name: rfgain
    gui_hint: 1,0,1,4
- name: trans_width
    gui_hint: 2,1,1,1
- name: tuner
    gui_hint: ''
- name: volume
    gui_hint: 1,1,1,1
- name: volume
    gui_hint: 1,5,1,8
- name: qtgui_freq_sink_x_0
    gui_hint: 3,0,15,2
```

jin-github1002, 2019-11-20:
[image]

i wanna achieve this. Is it ok that i set gain(2,3,1,1)?
I think i need to delete on volume, but i cant find two volume in the graph.
And how can i see the print info above that you listed, run it with some option？

Thank you so much！

velichkov, 2019-11-20:
> i wanna achieve this. Is it ok that i set gain(2,3,1,1)?

Yes, probably.

> I think i need to delete on volume, but i cant find two volume in the graph.

One of the `QT GUI Range` blocks with name `volume` is in the top right corner of your flowgraph but it is disabled, so you could ignore or delete it. The second `volume` block is in the bottom left corner.

> And how can i see the print info above that you listed, run it with some option？

Execute the following commands in your terminal

```
cd /home/dell/Desktop/
grep "name:\|gui_hint:" hello_world2.grc 
```
Note that I've deleted some of the output to make the problem clearer.

jin-github1002, 2019-11-21:
i changed. but the button can not click. see the 1st picture.(name is helloworld2.py)
[image]

Then i deleted all the GUI hint, the buttons can be clicked. But when using the tuner to change the fm frequency, there is no any voice. whats wrong?(name is test.py) I uploaded the 2nd picture and the output in the console.
[image]
[testInConsole.txt](https://github.com/gnuradio/gnuradio/files/3872703/testInConsole.txt)

velichkov, 2019-11-21:
> i changed. but the button can not click. see the 1st picture.(name is helloworld2.py)

What exactly have you changed?

> Then i deleted all the GUI hint, the buttons can be clicked. But when using the tuner to change the fm frequency, there is no any voice. whats wrong?(name is test.py) I uploaded the 2nd picture and the output in the console.

> ##################
> LimeSuite version: 19.04.1-ga5b3a10f
> gr-limesdr version: 3.0.0.0
> ##################

Which Linux distribution do you use and how did you install `gnuradio`, `LimeSuite` and `gr-limesdr`? 

From one of your [screenshots](https://user-images.githubusercontent.com/56423419/69269392-ea306580-0b9e-11ea-9777-fc1642e956cf.png) it looks like you are following a youtube video where the author is using gnuradio 3.7. Could you provide a link to this video and to the original XML based flowgraph that you probably are trying to port to gnuradio 3.8?

Why did you choose to use gnuradio 3.8? It's relatively new and probably there are still some bugs.

jin-github1002, 2019-11-21:
https://www.youtube.com/watch?v=2Nd9BIYmKD8
This is the tutorial video that i followed.

I heard gnuradio3.8 is a released version. Then i chose it. I think it is possible because of some periorites that i dindt set correctly.(e.g in LimeSDR Source, i set to use channel a and i set an antenna RX1-L).   i uploaded the grc file , py file and the hardware here. 

[test.zip](https://github.com/gnuradio/gnuradio/files/3876170/test.zip)
[image]
[image]

dkozel, 2019-12-06:
@jin-github1002 Were you able to get your application working?

For application assistance the GNU Radio mailing list or chat channel are the best place to get assistance. We try to keep the GitHub Issue tracker for monitoring errors or feature requests in the main code. I've transmitted and received WBFM using GNU Radio 3.8 so I'm going to close this issue, happy to help you either via the email list or chat.

https://wiki.gnuradio.org/index.php/MailingLists
https://wiki.gnuradio.org/index.php/IRC

closed 2019-12-06 by dkozel