gnuradio/gnuradio #2911: gr-blocks/stream_mux: Default value for int_vector causes error due to not being a vector

ThomasHabets, 2019-11-17:
The `lengths` arg defaults to `1,1`. That doesn't work. It works if the default is `[1, 1]`.

So should it:
a. Always default to `[1,1]` if it's an int_vector, or
b. Should `1,1` be accepted and understood to be a vector?

If (b), what would it mean if user provides `[1,1]`?

I haven't checked for other blocks with the same problem yet, since I want to know what pattern is right and which is wrong.

michaelld, 2019-11-17:
This was also on IRS #gnuradio ... my vote is to use `[1,1]` ... it just seems to be a "vector" to me, while `1,1` is not necessarily.

ThomasHabets, 2019-11-18:
It's also a bit odd that if you put `1,1` as length and the `make`-line is:

```
    make: blocks.stream_mux(${type.size}*${vlen}, ${lengths})
```

then it expands entirely texturally and will try to pass three arguments to `stream_mux()`. That may be more flexible, but is it intended? Even if we change the default to `[1,1]` the error message for `1,1` may be confusing.

michaelld, 2019-11-18:
That's true. We can't keep users from entering in invalid info, of course! If there's a reasonable way to add a check to make sure the entry is a vector -- and if not, then print something useful -- then I'd say to try for that as well as changing the default. They do go together!

marcusmueller, 2024-01-07:
Can't reproduce: 
[image]
https://gist.github.com/marcusmueller/d36204ace863203401b33cf1b049878a

builds and runs. So, fixed!

closed 2024-01-07 by marcusmueller