gnuradio/gnuradio #2912: gr_modtool dies while editing binary file

michael-roe, 2019-11-17:
So, I built gnuradio-3.9 from the source in github, gnuradio-companion seems to work, but gr_modtool newmod  dies like this:

**************************************
$ /usr/local/bin/gr_modtool newmod c4fm
Creating out-of-tree module in ./gr-c4fm...
Traceback (most recent call last):
  File "/usr/local/bin/gr_modtool", line 30, in <module>
    cli()
  File "/usr/lib/python3/dist-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/modtool/cli/base.py", line 149, in wrapper
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/modtool/cli/newmod.py", line 64, in cli
    run(self)
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/modtool/cli/base.py", line 168, in run
    module.run()
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/modtool/core/newmod.py", line 91, in run
    s = filetext.read()
  File "/usr/lib/python3.6/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf3 in position 1: invalid continuation byte
*******************

newmod.py seems to be doing a search and replace of "HOWTO" in every file it copies .. but some of the files it copies are compiler output in .pyc files, and the search and replace dies because the file contents aren't UTF-8.

michael-roe, 2019-11-19:
I'm not sure what went wrong here, but this was on a system where I had previously had gnuradio-3,7  installed.

The file $(HOME)/.gnuradio/config.conf sets newmod_path, which will be where the previously installed version of gnuradio got its template from.

mormj, 2019-11-20:
I've seen this before - thought it was due to a packaging quirk, but it looks like you're seeing it installed from source.  I think the issue is that __pycache__ dirs are installed into 

/usr/share/gnuradio/modtool/templates/gr-newmod

running 
```
sudo py3clean -v /usr/share/gnuradio/modtool/templates/gr-newmod
```

Will work around the issue for now, but a solution in the install scripts still needs to be figured out

mormj, 2020-04-09:
This has been fixed in the debian builds and in the ubuntu ppa builds.  Closing.

closed 2020-04-09 by mormj