gnuradio/gnuradio #2915: GR 3.8 OOT module fails with log4cpp dependency

mwk088, 2019-11-20:
gnu:  3.8-maint
os:  CentOS 8

I am trying to build and OOT module and I am getting the error seen below.

Any ideas?  Do I add linker flags in my cmake command?

Thank you

Mark

`#15 44.86 [ 93%] Linking CXX executable test_eventstream
#15 44.86 cd /workspaces/ta/gr-eventstream/Release/lib && /usr/bin/cmake -E cmake_link_script CMakeFiles/test_eventstream.dir/link.txt --verbose=ON
#15 44.88 /usr/bin/c++  -fpermissive -g  -rdynamic CMakeFiles/test_eventstream.dir/qa_es.cc.o CMakeFiles/test_eventstream.dir/qa_es_common.cc.o CMakeFiles/test_eventstream.dir/qa_es_source.cc.o CMakeFiles/test_eventstream.dir/qa_es_sink.cc.o CMakeFiles/test_eventstream.dir/qa_es_trigger.cc.o CMakeFiles/test_eventstream.dir/test_all.cc.o  -o test_eventstream -Wl,-rpath,/workspaces/ta/gr-eventstream/Release/lib libeventstream.so -lcppunit -ldl -lboost_system -lboost_date_time -lboost_program_options -lboost_thread -lboost_regex -lboost_chrono -lboost_atomic -lgnuradio-runtime -lgnuradio-blocks -lgnuradio-pmt -lpython3.6m 
**#15 45.11 /usr/bin/ld: CMakeFiles/test_eventstream.dir/qa_es_common.cc.o: undefined reference to symbol '_ZN7log4cpp8Appender29AppenderMapStorageInitializerC1Ev'
#15 45.11 /opt/truearrow/6.3/lib/liblog4cpp.so.5: error adding symbols: DSO missing from command line**
#15 45.11 collect2: error: ld returned 1 exit status
#15 45.11 make[2]: *** [lib/CMakeFiles/test_eventstream.dir/build.make:172: lib/test_eventstream] Error 1
#15 45.11 make[2]: Leaving directory '/workspaces/ta/gr-eventstream/Release'
#15 45.11 make[1]: *** [CMakeFiles/Makefile2:120: lib/CMakeFiles/test_eventstream.dir/all] Error 2
#15 45.11 make[1]: *** Waiting for unfinished jobs....
#15 67.20 [ 95%] Linking CXX shared module _es_swig.so
#15 67.20 cd /workspaces/ta/gr-eventstream/Release/swig && /usr/bin/cmake -E cmake_link_script CMakeFiles/_es_swig.dir/link.txt --verbose=ON
#15 67.23 /usr/bin/c++ -fPIC -fpermissive -g  -shared  -o _es_swig.so CMakeFiles/_es_swig.dir/es_swigPYTHON_wrap.cxx.o -Wl,-rpath,/workspaces/ta/gr-eventstream/Release/lib: -lpython3.6m ../lib/libeventstream.so -lcppunit -ldl -lboost_system -lboost_date_time -lboost_program_options -lboost_thread -lboost_regex -lboost_chrono -lboost_atomic -lgnuradio-runtime -lgnuradio-blocks -lgnuradio-pmt -lpython3.6m 
#15 67.92 make[2]: Leaving directory '/workspaces/ta/gr-eventstream/Release'
#15 67.94 [100%] Built target _es_swig
#15 67.94 make[1]: Leaving directory '/workspaces/ta/gr-eventstream/Release'
#15 67.94 make: *** [Makefile:141: all] Error 2
#15 67.95 [ Build gr-eventstream (Release) FAILED with error code 2 ]
#15 ERROR: executor failed running [/bin/sh -c ./scripts/build_cmake_project.sh gr-eventstream Release;]: exit code: 2
------
 > [11/18] RUN ./scripts/build_cmake_project.sh gr-eventstream Release;:
------
failed to solve with frontend dockerfile.v0: failed to build LLB: executor failed running [/bin/sh -c ./scripts/build_cmake_project.sh gr-eventstream Release;]: exit code: 2
The terminal process terminated with exit code: 1`

velichkov, 2019-11-20:
Hi @mwk088,

> gnu: 3.8-maint
> os: CentOS 8

It would be very helpful if you could share all the dockerfiles, shell scripts and spec files you are using to build gnuradio 3.8 and the other OOT modules on CentOS 8. And for reference a [link to your malling list post](https://lists.gnu.org/archive/html/discuss-gnuradio/2019-11/msg00194.html)

> I am trying to build and OOT module and I am getting the error seen below.
> 
> Any ideas? Do I add linker flags in my cmake command?

> ## `#15 44.86 [ 93%] Linking CXX executable test_eventstream
> #15 44.86 cd /workspaces/ta/gr-eventstream/

Which `gr-eventstream` version are you trying to compile and from which repository?

The `gr-eventstream` from it's [upstream repo](https://github.com/osh/gr-eventstream) has not yet been ported to gnuradio 3.8. 

There is a `maint-3.8` branch in [this fork](https://github.com/flynn378/gr-eventstream/tree/maint-3.8) which I was able to compile with gnuradio 3.9 (the master branch) on Fedora 29. So try building from this fork and branch. 

> #15 44.88 /usr/bin/c++ -fpermissive -g -rdynamic CMakeFiles/test_eventstream.dir/qa_es.cc.o CMakeFiles/test_eventstream.dir/qa_es_common.cc.o CMakeFiles/test_eventstream.dir/qa_es_source.cc.o [...]

With gnuradio 3.8 and 3.9 the `log4cpp` is added to the linker flags by these two cmake files

```
cmake/gnuradio/gnuradio-pmtTargets.cmake:  INTERFACE_LINK_LIBRARIES "Boost::boost;Boost::thread;Log4Cpp::log4cpp"
cmake/gnuradio/gnuradio-runtimeTargets.cmake:  INTERFACE_LINK_LIBRARIES "gnuradio::gnuradio-pmt;Volk::volk;Boost::program_options;Boost::filesystem;Boost::system;Boost::regex;Boost::thread;Log4Cpp::log4cpp;MPLib::mplib;rt"
```
and when you link your library with either `gnuradio-runtime` or `gnuradio-pmt` the `log4cpp` gets added to the linker flags together with the other libraries.

P.S.
You need to enclose the logs in triple back quotes (```), see https://guides.github.com/features/mastering-markdown/#GitHub-flavored-markdown

mwk088, 2019-11-20:
Vasil,
Thank you for the suggestions.  I am using a slightly modified version of gr-evenstream, however, I will look to build with the one you provided a link.

I also appreciate the tips on how to effectively paste in code.

Mark

dkozel, 2019-12-06:
@mwk088 were you able to solve your issue?

mwk088, 2019-12-06:
No, but I was briefly redirected to another task.  I will be continuing this task soon, and will have more questions and information to share.

thanks.

dkozel, 2019-12-06:
You're welcome. When you do jump back on this task can you please post to the mailing list or chat channel? They are the prefered, and more effective, places to solve install and application issues. Feel free to tag me (dkozel or Derek Kozel) on chat.

https://wiki.gnuradio.org/index.php/IRC
https://wiki.gnuradio.org/index.php/MailingLists

closed 2019-12-06 by dkozel

mwk088, 2019-12-12:
I was able to build my OOT module and get pass the liblog4cpp error by doing the following:  using the following patch and getting the updated FindLOG4CPP.cmake from the gnuradio master branch

```diff --git a/lib/CMakeLists.txt b/lib/CMakeLists.txt
index 65d67daea9..42c6926953 100644
--- a/lib/CMakeLists.txt
+++ b/lib/CMakeLists.txt
@@ -59,7 +59,7 @@ add_library(eventstream SHARED ${eventstream_sources})
 target_link_libraries(eventstream ${eventstream_libs}) add_executable(test_eventstream ${test_eventstream_sources} test_all.cc)
-target_link_libraries(test_eventstream eventstream ${CPPUNIT_LIBRARIES})
+target_link_libraries(test_eventstream eventstream ${CPPUNIT_LIBRARIES} `${LOG4CPP_LIBRARIES})```