gnuradio/gnuradio #2921: Is separate visualization for different blocks (only complex ones) possible with performance counters enabled in GNURadio 3.7.13.

shrin18, 2019-11-22:
Hello,

I am running GNURadio 3.7.13 with performance counters enabled in Ubuntu 18.04. Thrift version is 0.10.0 and it works perfectly fine with the graphs on the gr-perfx-monitor. However, since we have many blocks in the flowgraph and some blocks have a hierarchical block inside them, it is difficult for us to view the runtime measurements and the graphs for percentage full. 

We wanted to know whether there is any way to get the profiling parameters for certain blocks only and not all ?. We have placed the control probe on only the block that we want but still the performance monitor gives the results for all the blocks. Visualising the counters is getting difficult. See below !

We want to measure only the work taken by the complex blocks such as the decoding block in our case where we are decoding a cdma signal from a uhd source and getting the output at one of the hier blocks as a decoded payload.

[image]
[image]

Please let me know is there any way to do this.
Thanks,
Shrinish Donde

michaelld, 2019-11-25:
Hi Shrinish - Shortest answer is: I don't think so. If Tom doesn't mention it, then it probably doesn't exist. It would be an excellent feature to add to the codebase if it truly doesn't exist yet. That and updating the codebase to work with more recent Thrift would be desirable. Not sure if those are within your domain of expertise and time, but they would be great to have!

shrin18, 2019-11-25:
We stuck to the workaround using htop and enabling the custom thread names options in the display. We could integrate the PID of each block separately and recorded it using linux perf tools. 

I would be happy to have a look at the code for the (gr-perfx-monitor) and try to incorporate the changes similarly, since each block is running on a separate thread. If sorting of blocks is possible, then so must be this. I will get back when I have a reasonable PR ready.

Thanks for the suggestion !

marcusmueller, 2020-01-06:
Dear Shrinish,

since 3.7.x is on legacy / maintenance mode, we're not adding any features to it any more to avoid maintainer overhead (we're very manpower-bounded).

Generally, if you're planning to add new features, I'd very much recommend doing so against master – that's our forward-facing branch. You'd get *a lot* of benefits from moving forward from GNU Radio 3.7; Python3, for example.

We're obviously willing to accept smaller patchsets for 3.7, if they are also forward-portable to master, and do not break 3.7 ABI compatibility (i.e. we can't accept features that require newer versions of Thrift).

marcusmueller, 2024-01-08:
Seeing how little love ctrlport and perfcounters have gotten in the last 8 years… I'm not giving this much of a chance of really happening, unless someone gets the shuffles and starts redoing all of ctrlport.

closed 2024-01-08 by marcusmueller