gnuradio/gnuradio #2924: [gnuradio/gnuradio] how to reference another OOT's swig in an OOT's python test

derekjchen, 2019-11-24:
I wrote a dvbs2 bbheader file sink. (for learning purpose). I did cmake, make, make install, ldconfig on it.

and I'm writing python tests for dvbs2 bbheader using the bbheader file sink block.
however, in the python test file, qa_bbheader_bb.py, it can’t resolve this:
import bbheader_file_sink_swig as bbheader_file_sink

2: Test command: /usr/bin/sh "/data/Frontier/gr-dvbs2/build/python/qa_bbheader_bb_test.sh"
2: Test timeout computed to be: 10000000
2: Traceback (most recent call last):
2:   File "/data/Frontier/gr-dvbs2/python/qa_bbheader_bb.py", line 27, in <module>
2:     import bbheader_file_sink_swig as bbheader_file_sink
2: ImportError: No module named bbheader_file_sink_swig
2/8 Test #2: qa_bbheader_bb ...................***Failed    0.25 sec
test 3
  Start 3: qa_bbscrambler_bb

and I can see bbheader_file_sink is under:
/user/local/include/bbheader_file_sink/bbheader_file_sink/swig/bbheader_file_sink_swig.i

should I edit dvbs2 OOT CMakeFile? and how?

Really appreciate any help.

michaelld, 2019-11-25:
Did you install the OOT already? If so, where is the file "_ bbheader_swig.so" installed? Guessing it will be in a directory that matches `/usr/local/lib/python*/*packages/bbheader/`... but that's just a guess.

michaelld, 2019-11-25:
If you haven't installed it yet, then you'll want to do the following to get the "make test" to work:
{{{
cd /data/Frontier/gr-dvbs2/python/
ln -s bbheader .
}}}

michaelld, 2019-11-25:
And now I'm wondering why the directory is called `gr-dvbs2` where the files are being placed? Did you do this work inside the `gr-dvbs2` submodule in GR? Or in some OOT?

derekjchen, 2019-11-25:
Hi Michael, 
thanks for the reply.
yes, _bbheader_file_sink_swig.so is installed here:
/usr/local/lib/python2.7/dist-packages/bbheader_file_sink/_bbheader_file_sink_swig.so

yes, gr-dvbs2 is the OOT I want to reference bbheader_file_sink from.

sorry, I did those:
```
cd /data/Frontier/gr-dvbs2/python/
ln -s bbheader .
```
and I still have the same error.

anything else I could try?

regards,
Derek

michaelld, 2019-11-25:
Ah; OK. So you're OOT is actually called `bbheader_file_sink`, not just `bbheader`.

Inside some other GR OOT, you should be able to do `import bbheader_file_sink` directly.

You can remove the link here then `/data/Frontier/gr-dvbs2/python/bbheader`.

derekjchen, 2019-11-26:
Hi Michael,
thanks for your reply!
I see... I should have just imported bbheader_file_sink directly, not the bbheader_file_sink_swig ...

it's working fine now.

thanks again.

closed 2019-11-26 by derekjchen