gnuradio/gnuradio #2925: Curved arrows outside the bounding box

michael-roe, 2019-11-25:
If I use file->Screen Capture in gnuradio-companion (master branch), and the right-most part of the flowgraph is an arrow that leaves the right-hand side of a block and enters the left-hand side of a block, the curved arrow goes off the edge of the PDF that is created. (I guess something is wrong with the bounding box computation for curved arrows)

The flowgraph is like this:

Block  ---+
Block <--+

mormj, 2019-11-27:
I think I understand, but any way you can throw a screenshot on here for clarification?

haakov, 2020-01-04:
I was able to reproduce:

[image]

marcusmueller, 2024-01-08:
[image]

yep, still happening ( https://gist.github.com/marcusmueller/c98f4e95d6fb2f7fb9f23921bf6511f8
 for flowgraph)

marcusmueller, 2024-03-11:
tagging low because of qt-grc