gnuradio/gnuradio #2926: Additive scrambler reset by count

michael-roe, 2019-11-25:
Additive scrambler seems to work OK with a tagged stream. But if a count rather than a tag is used to reset it, it looks like the sequence gets out of step.

Here's a dump of the output with mask=011,seed=0x1c9,length=8,count=8,bits_per_byte=1,reset_tag_key=""

contents = 
0000: 01 00 00 01 00 00 01 01 01 01 00 00 01 00 00 01 
0010: 01 01 00 00 01 00 00 01 01 01 00 00 01 00 00 01 
0020: 01 01 00 00 01 00 00 01 01 01 00 00 01 00 00 01 
0030: 01 01 00 00 01 00 00 01 01 01 00 00 01 00 00 01 
0040: 01 01 00 00 01 00 00 01 01 01 00 00 01 00 00 01 

The first 8 values are OK, and then (after the first reset) it gets out of step.

(The particular LFSR above is the same as the one used in Yaesu System Fusion. In real use, frames would be longer than 8 bits but I'm using 8 bit frames to show the bug more easily),

michael-roe, 2019-11-25:
P.S. This was  with gnuradio built from the master branch.

rear1019, 2019-12-30:
There is indeed an error in ``additive_scrambler_bb_impl.cc``: Every time ``work()`` is called the very first reset is delayed by one byte (the reset is byte based). I might make a pull request with a fix by end of the week.

commit fddb0d4e (2020-01-04): additive_scrambler: Fix count based reset

Count based reset in the additive scrambler may be delayed by a single
byte as reported in [1]. This happens whenever the work() function of
the block is called: The first value of reset_index in work() is d_count
- d_bytes where d_count is the number of bytes to output before a reset
and d_bytes is the number of bytes produced after a previous reset (in
the previous work() call). The very first reset is delayed by one byte
due to zero-based counting in work()’s loop /and/ a check of the reset
condition (i == reset_index) /after/ outputting a byte. (A [... cut]
commit 9142c1ea (2020-01-05): additive_scrambler: Fix count based reset

Count based reset in the additive scrambler may be delayed by a single
byte as reported in [1]. This happens whenever the work() function of
the block is called: The first value of reset_index in work() is d_count
- d_bytes where d_count is the number of bytes to output before a reset
and d_bytes is the number of bytes produced after a previous reset (in
the previous work() call). The very first reset is delayed by one byte
due to zero-based counting in work()’s loop /and/ a check of the reset
condition (i == reset_index) /after/ outputting a byte. (A [... cut]
commit 2d7f8234 (2020-01-16): additive_scrambler: Fix count based reset

Count based reset in the additive scrambler may be delayed by a single
byte as reported in [1]. This happens whenever the work() function of
the block is called: The first value of reset_index in work() is d_count
- d_bytes where d_count is the number of bytes to output before a reset
and d_bytes is the number of bytes produced after a previous reset (in
the previous work() call). The very first reset is delayed by one byte
due to zero-based counting in work()’s loop /and/ a check of the reset
condition (i == reset_index) /after/ outputting a byte. (A [... cut]
closed 2020-01-04 by mbr0wn